cartographer-project/cartographer_ros #1351: Clip on z-axis for 2D map projection

JulianKu, 2019-07-09:
Hi,

I am doing indoor 3D SLAM using a [Livox Mid-100](https://www.livoxtech.com/mid-40-and-mid-100/specs) Lidar sensor and I am getting rather good 3D maps (visualized with the point cloud viewer).
However, I would like to use projected 2D xray maps for autonomous navigation on this map. I couldn't find any way to specify the range along the z-axis for which the projection should happen. By this I would want to incorporate only obstacles that are within the height of my autonomous vehicle into the map and e.g. not bridges where the robot could pass underneath easily. Did I miss something, is there a way?

Clipping the point cloud itself during/before performing SLAM is a bad option in my opinion, as I am losing "features" for a better/more accurate map creation.

dhordiiuk, 2020-05-30:
hey @JulianKu ! Could you please share the way you did 3D maps using  https://www.livoxtech.com/mid-40-and-mid-100 ? Unfortunnatelly I'm not successed in that till today

JulianKu, 2020-05-30:
Hi,
You can have a look at the "devel_slam" branch of our repository https://github.com/Roboy/autonomous_driving_v2/blob/master/README.md. See the attached wiki for further reference.
We did succeed for our scenarios. Used the Livox Mid 100 in combination with SBG Ellipse IMU

dhordiiuk, 2020-05-30:
@JulianKu thanks a lot!

closed 2020-10-16 by MichaelGrupp

fjc378209410, 2021-01-14:
Hi,
    I am trying to combine livox and cartographer for slam, but my livox_frame keeps beating, how did you solve this problem?

BIRL-xu, 2021-11-11:
Hi @JulianKu, I also want to to the 2D occupancy grid map from point coud map, do you have a solution?

JulianKu, 2021-11-11:
@BIRL-xu  You can project your 3D Pointcloud to a 2D plane using the Cartographer Assets Writer and then save this projection to an image (see [doc here](https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html)). To load as 2D occupancy grid. 
Then you can pass this image to the [ROS map server](http://wiki.ros.org/map_server) which publishes this as an occupancy grid.

BIRL-xu, 2021-11-12:
@JulianKu,cool!, I got the occupancy grid map. Thank you very much!
[image]

> @BIRL-xu You can project your 3D Pointcloud to a 2D plane using the Cartographer Assets Writer and then save this projection to an image (see [doc [...]