cartographer-project/cartographer_ros #1357: loop closing failure on Deutsches museum 3d demo

babiking, 2019-07-12:
I'm trying to run a 3D demo on Deutsches museum data.

I fetch the master branch of cartographer_ros (hash: 20c2c8f0f7df5d36828fc3af3ad4b63687adb388) and cartographer (hash: bcd5486025df4f601c3977c44a5e00e9c80b4975) and launch cartographer_node by command: "roslaunch cartographer_ros demo_backpack_3d.launch bag_filename:=./b3-2016-04-05-14-14-00.bag", without any modification on configuration / launch / urdf files. However, the output of global SLAM seems unreasonable. At the loop closing positions, few INTER constraints were computed and submap poses hadn't been globally optimized (as shown in following figure).

[image]

would anyone provide some instructions or hints to debug? Thanks a lot :)

aidamQin, 2019-08-09:
please elevate your CPU level

yangtfu, 2020-02-13:
same result

luanjinlu, 2020-04-02:
I also have this result, do you have  answer? @babiking

eithwa, 2020-10-31:
https://github.com/cartographer-project/cartographer_ros/blob/70612a4d40f684276e3185c3499fc56faf64201c/cartographer_ros/configuration_files/backpack_3d.lua#L53

POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 20
[image]
[image]

MichaelGrupp, 2020-11-09:
@eithwa Nice, did you test it with the master branch and check if this works well with multiple of the 3D backpack bagfiles? If yes, it would be nice if you could open a pull request so we can change this parameter, and maybe others too if needed.