cartographer-project/cartographer_ros #1358: Pure Localization Can Update/Expand Map?

pephabet, 2019-07-15:
I've been doing some experiments using Pure Localization and I've noticed that not only does it do a great job of localizing the robot inside a previously created map, but it also updates and expands the map. Now I'm curious what are the differences between running Cartographer and running Pure Localization (aside from the fact that Pure Localization can load a previously saved map). Because to me it seems, unlike then name suggests, Pure Localization is also performing SLAM.

robot-mh, 2019-08-12:
can not update map, only reserved 3~5 submaps and the map before is not particapate optimization with this trajectory

MichaelGrupp, 2020-10-27:
Pure localization also does SLAM, it just keeps a limited history of submaps (depending on the configuration) to bound memory usage.

closed 2020-10-27 by MichaelGrupp