cartographer-project/cartographer_ros #1359: error with "Queue waiting for data:(0, points2)

CherryChenw, 2019-07-16:
Hello!
I have a 3d LiDAR bag containing a pointCloud2 /fullscan msg and a IMU /imu/data msg.
I tested cartographer ROS with this bag.

The configuration files , launch file and urdf file is here
https://github.com/minmummax/ACTL/tree/master/cartographer

Note that in .launch,
I have been remapped:
remap from="points2" to="/rslidar_points" /
remap from="imu" to="/imu" /

In .urdf ,
The frame id of "/rslidar_points" is "rslidar"
The frame id of "/imu" is "imu_link"

Here is the log:
nvidia@tegra-ubuntu:~$ roslaunch cartographer_ros my_robot.launch 
... logging to /home/nvidia/.ros/log/feb19eee-a78e-11e9-926b-00044bc4c36f/roslaunch-tegra-ubuntu-6029.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:34058/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [6046]
process[cartographer_node-2]: started with pid [6047]
process[cartographer_occupancy_grid_node-3]: started with pid [6054]
[ INFO] [1563260714.047511330]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer_ros/configuration_files/my_robot.lua' for 'my_robot.lua'.
[ INFO] [1563260714.048280264]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1563260714.048386696]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1563260714.048548170]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1563260714.048641322]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1563260714.048952493]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1563260714.049032973]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1563260714.049163662]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1563260714.049242639]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1563260714.049475697]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1563260714.049570801]: I0716 15:05:14.000000  6047 configuration_file_resolver.cc:41] Found '/home/nvidia/cherrychen/catkin_cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1563260714.098985099]: I0716 15:05:14.000000  6047 submap_3d.cc:321] Added submap 1
[ INFO] [1563260714.099301486]: I0716 15:05:14.000000  6047 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1563260718.614674380]: I0716 15:05:18.000000  6047 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636988549535220397'.
[ WARN] [1563260725.693185513]: W0716 15:05:25.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260726.400016860]: W0716 15:05:26.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260727.100997989]: W0716 15:05:27.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260727.801131980]: W0716 15:05:27.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260728.583783349]: W0716 15:05:28.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260729.310638768]: W0716 15:05:29.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260729.990800548]: W0716 15:05:29.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260730.701135623]: W0716 15:05:30.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
[ WARN] [1563260731.393236095]: W0716 15:05:31.000000  6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)

The ROS report an error with "Queue waiting for data:(0, points2)" when I try to test catographer with bag， and in RVIZ there is no data.
here is my bagtest:
[... 10 lines cut ...]
[0.000299, 0.029360)	####################	Count: 41276 (98.699188%)	Total: 41276 (98.699188%)
[0.029360, 0.058422)	                    	Count: 396 (0.946915%)	Total: 41672 (99.646103%)
[0.058422, 0.087484)	                    	Count: 72 (0.172166%)	Total: 41744 (99.818268%)
[0.087484, 0.116546)	                    	Count: 37 (0.088474%)	Total: 41781 (99.906746%)
[0.116546, 0.145607)	                    	Count: 24 (0.057389%)	Total: 41805 (99.964134%)
[0.145607, 0.174669)	                    	Count: 8 (0.019130%)	Total: 41813 (99.983261%)
[0.174669, 0.203731)	                    	Count: 5 (0.011956%)	Total: 41818 (99.995216%)
[0.203731, 0.232792)	                    	Count: 1 (0.002391%)	Total: 41819 (99.997612%)
[0.232792, 0.261854)	                    	Count: 0 (0.000000%)	Total: 41819 (99.997612%)
[0.261854, 0.290916]	                    	Count: 1 (0.002391%)	Total: 41820 (100.000000%)
E0716 14:32:23.135694  5234 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.114516 s, recommended is [0.0005, 0.05] s with no jitter.
I0716 14:32:23.135882  5234 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 4549  Min: 0.086881  Max: 0.114516  Mean: 0.100800
[0.086881, 0.089644)	                    	Count: 2 (0.043966%)	Total: 2 (0.043966%)
[0.089644, 0.092408)	                    	Count: 1 (0.021983%)	Total: 3 (0.065949%)
[0.092408, 0.095171)	                    	Count: 3 (0.065949%)	Total: 6 (0.131897%)
[0.095171, 0.097935)	                    	Count: 11 (0.241811%)	Total: 17 (0.373709%)
[0.097935, 0.100699)	                  ##	Count: 444 (9.760386%)	Total: 461 (10.134095%)
[0.100699, 0.103462)	  ##################	Count: 4069 (89.448227%)	Total: 4530 (99.582329%)
[0.103462, 0.106225)	                    	Count: 13 (0.285777%)	Total: 4543 (99.868103%)
[0.106225, 0.108989)	                    	Count: 3 (0.065949%)	Total: 4546 (99.934052%)
[0.108989, 0.111752)	                    	Count: 1 (0.021983%)	Total: 4547 (99.956032%)
[0.111752, 0.114516]	                    	Count: 2 (0.043966%)	Total: 4549 (100.000000%)

Thank you!

CherryChenw, 2019-07-16:
I am uploading my bag，please wait

CherryChenw, 2019-07-16:
here is my bag：https://pan.baidu.com/s/1IUQN4KcPNorohTK4bhk_KQ  
Extraction code：a0gh

UserUsingGit, 2019-07-18:
Hi, as far as I see there is a link to your urdf file missing in my_robot.launch.
<param name="robot_description" textfile="$(find cartographer_ros)/urdf/test.urdf" />
points to "test.urdf", but it should point to "my_robot.urdf". I do not know if that solves the error but you might give it a try.

MrGitGo, 2019-07-23:
This: "W0716 15:05:31.000000 6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)" indicates that cartographer does not get any pointclouds. So there is something wrong with your pointcloud. Check if it is publishing point cloud or point cloud2. Check with rotopic echo <topic_name> if you get any point cloud data

CherryChenw, 2019-07-23:
> This: "W0716 15:05:31.000000 6047 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)" indicates that cartographer does not get any pointclouds. So there is something wrong with your [...]

when I run rviz and play this bag,I can see like this:
[image]

MrGitGo, 2019-07-23:
Ok, first of all you should fix your tf tree. I guess the warning comes because you don't have an odom tf. So try to set  provide_odom_frame  to true, and published_frame to "odom".

CherryChenw, 2019-07-23:
> Ok, first of all you should fix your tf tree. I guess the warning comes because you don't have an odom tf. So try to set provide_odom_frame to true, and published_frame to "odom".

I change my lua file.

options = {
 map_builder = MAP_BUILDER,
 trajectory_builder = TRAJECTORY_BUILDER,
 map_frame = "map",
 tracking_frame = "imu",
 published_frame = "odom",
 odom_frame = "odom",
 provide_odom_frame = true,
 publish_frame_projected_to_2d = false,
 use_odometry = false,
 use_nav_sat = false,
 use_landmarks = false,
 num_laser_scans = 0,
 num_multi_echo_laser_scans = 0,
 num_subdivisions_per_laser_scan = 1,
 num_point_clouds = 1,
 lookup_transform_timeout_sec = 0.2,
 submap_publish_period_sec = 0.3,
 pose_publish_period_sec = 5e-3,
 trajectory_publish_period_sec = 30e-3,
 rangefinder_sampling_ratio = 1.,
 odometry_sampling_ratio = 1.,
 fixed_frame_pose_sampling_ratio = 1.,
 imu_sampling_ratio = 1.,
 landmarks_sampling_ratio = 1.,
}

and I can't see any log when bag is playing.
[image]

CherryChenw, 2019-07-23:
> Ok, first of all you should fix your tf tree. I guess the warning comes because you don't have an odom tf. So try to set provide_odom_frame to true, and published_frame to "odom".

I think it doesn't work.

CherryChenw, 2019-07-23:
> Ok, first of all you should fix your tf tree. I guess the warning comes because you don't have an odom tf. So try to set provide_odom_frame to true, and published_frame to "odom".

I reboot my pc.and the question is the same:Queue waiting for data: (0, points2)
I have play some bag which record by others, it works correctly. so ,I think there is some thing wrong with my bag ,but I don't know what it is.

Sumyer, 2019-10-23:
@CherryChenw 
Hello！
 I have the same problem as you.
Have you solved it?

saharLeisiazar, 2019-11-06:
Hi
could you solve your problem?

starterpeng, 2020-08-18:
Sadlly, I meet the same problem. Did you find any solutions? I guess this problem indicates that the whole algorithm cant not compute the "matched scan points" instead of can not receive  the input point cloud. But I don't know how it happened, everything out of that looks fine......

yuan608, 2020-10-14:
Hi
I also have the similar issue.
 [ WARN] [1602663416.026294675, 1432647608.303914625]: W1014 16:16:56.000000 20285 ordered_multi_queue.cc:155] Queue waiting for data: (0, echoes)
Have you solved it?

MichaelGrupp, 2020-10-16:
This can be very likely solved by remapping the topic names. See also https://github.com/cartographer-project/cartographer_ros/issues/1414#issuecomment-710109412

closed 2020-10-16 by MichaelGrupp