cartographer-project/cartographer_ros #1363: Pass an argument into lua file

realdealneil, 2019-07-18:
I hope this is an easy issue for someone that knows lua better than me.  We have a multi-robot system where we desire to instantiate robots with various names and have each of them run cartographer.  Right now, the way we do this is by calling a ros launch file like the following:

<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<launch>
  <arg name="name" />
  <arg name="mode" default="3d"/>
  

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
      -configuration_directory $(find ssci_control)/config/cartographer
      -configuration_basename cartographer_$(arg mode)_$(arg name).lua"
      output="screen" required="true">
      <remap from="scan" to="front_scan" />
      <remap from="points2" to="points" />
      <remap from="imu" to="imu/data"/>
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
    type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
</launch>

We pass in the name of the vehicle and the mode (2d or 3d).  Then, we select the appropriate name for the lua file from that information.  The lua file then calls a generic lua file with a make_options('name') function in it, passing in the name so the lua file can construct the right topic names/tf frame names.  However, we have to make lua files for each different robot name in order to make this work.  Is it possible to pass an argument to the lua configuration file through the launch file?  I can't quite figure out how to do this.

realdealneil, 2019-07-18:
Third time's the charm?  Here's the relevant portion of the launch file:

node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
      -configuration_directory $(find ssci_control)/config/cartographer
      -configuration_basename cartographer_$(arg mode)_$(arg name).lua"
      output="screen"

AlexReimann, 2019-07-19:
Use triple ` ` ` for [markdown](https://guides.github.com/features/mastering-markdown/) formatting:
```
```xml <text here>```
```
We use a workaround with environment variables. You can [set envs in your launch] files which are scoped to this launch file:

```xml
<env name="MY_ROBOT" value="Robert">
```
You can then read the env in the lua file and do things accordingly:
```lua
robot_type = os.getenv("MY_ROBOT")

if robot_type == "Robert" then
  options.use_odometry = false
end
```