cartographer-project/cartographer_ros #1367: what is difference between inter & intra constraints

JaredLee1218, 2019-07-22:
the document said 
"Non-global constraints (also known as **inter submaps constraints**) are built automatically between nodes that are closely following each other on a trajectory. Intuitively, those “non-global ropes” keep the local structure of the trajectory coherent.

Global constraints (also referred to as loop closure constraints or **intra submaps contraints**) are regularly searched between a new submap and previous nodes that are considered “close enough” in space (part of a certain search window) and a strong fit (a good match when running scan matching). Intuitively, those “global ropes” introduce knots in the structure and firmly bring two strands closer."

so is it right that 
 inter constraints make submaps connected?
what is intra constraints role? help me plz...

nhphuong91, 2019-07-30:
If you know how to run turtlebot3 slam simulation example, try to look for an option name constraints on left side which belong to cartographer group and tick on it. My guess is that this is a visualization of what is inter submaps constraints. To put it in words, they are constraints between nodes which construct 1 submap. Node is a new scan that are different enough to be added to submap.
Therefore on the contrary, intra submaps contraints are links between new submap to nodes surround it but belong to other submaps.

Ellon, 2019-07-31:
_Disclaimer: The following comes from my understanding of the code. One of the developers may confirm if this is a good definition of constraints._

A constraint represents the estimated pose of the node j relative to the submap i. They can be differentiated between **intra**-submap (where node j was inserted into submap i) and **inter**-submap constraints (where node j was NOT inserted into submap i).

Basically, the **intra** constraints compose the "backbone" of the trajectory, while **inter** constraints represents loop closings in one trajectory or the joining of different trajectories (including the fixed one that should be supplied when running in pure localization mode).

I think the documentation is not consistent on this point and should probably be fixed.

MichaelGrupp, 2020-10-16:
The provided answer is correct. If you think portions of the docs can be improved, we are happy to receive a PR with your suggestions.

closed 2020-10-16 by MichaelGrupp

nhphuong91, 2020-10-16:
@MichaelGrupp Sorry for being too careful but can you tell which answer is correct? My explanation comes from a self-understanding and I haven't verify it with anyone in cartographer contributors group. Thanks in advanced!

MichaelGrupp, 2020-10-17:
No problem. I was referring to @Ellon's answer.