cartographer-project/cartographer_ros #1368: Map drifts after long rotation

ArcticCat, 2019-07-22:
Hello,
I'm using Google Cartographer (**master** branch) to build a 2D map with odometry and laser scan from RPLidar - A3 on Intel NUC with Intel Core i7-7567U and 16GB on Ubuntu 16.04 with ROS Kinetic. But map drifts while robot is rotated for a long time as you can see on images below:
[image]
[image]
It happens on angular velocity equals to 1 rad/sec. For a smaller velocities (near 0.2-0.3) map seems good.
I'm also using laser filter (from ROS laser_filters package : LaserScanAngularBoundsFilterInPlace) because of robot construction
[image]
I've tried to tune cartographer's local slam (POSE_GRAPH.optimize_every_n_nodes is set to 0), but could not achieve any good results.
This is my configuration file:  [configuration lua](https://github.com/ArcticCat/cartographer_data/blob/master/robot_lds_2d.lua)
There are my bag files (for first and second images respectively):
[bag of fist map](https://github.com/ArcticCat/cartographer_data/blob/master/data_rotating.bag)
[bag of second map](https://github.com/ArcticCat/cartographer_data/blob/master/data_moving.bag)
And there are output from rosbag_validate:
[rosbag_validate of first map's bagfile](https://github.com/ArcticCat/cartographer_data/blob/master/cartographer_validate_rotating.txt)
[rosbag_validate of second map's bagfile](https://github.com/ArcticCat/cartographer_data/blob/master/cartographer_validate_moving.txt)
There is also some problem with timestamps of `/odom` topic from OpenCR, so i change stams in another ros node (just using ros::Time::now() )and then send it to cartographer node via `remap from="/odom" to "/odom_server".`
So the question is:
**How can i tune Google Cartographer for my robot? Or the problem grows from robot's hardware and sensors?**

MrGitGo, 2019-07-23:
Change the rotation weight and look if it helps

ArcticCat, 2019-07-23:
> Change the rotation weight and look if it helps

There are modified weights in the configuration file. The problem appears after scan goes with it rear side split. Real time correlative scan matcher gives better results but they aren't ideal too. 

I'd like to understand if such scan filter is a problem for the Cartographer. Or maybe problem in scan rate or something else?

nhphuong91, 2019-07-30:
I'm dealing with the exact same problem as you. My temporary fix for it is to change ceres_scan_matcher.rotation_weight = 4e3, voxel_filter_size = 0.01 and use a phone as IMU source instead of regular MPU6050 (or similar). But as the map get bigger, it broken. Another problem is that this method seem to overwrite old/added scan within submap.
The rest setting is mostly same as turtlebot3 slam simulation example.
Tell me if you found anything else! I'm banging my head with the tuning instruction because it lack of image or video for illustration. Besides it does not tell when the tune is good enough to move on to the next tuning step so I'm just blindly tuning what I suspect to be the culprit right now.

Yu-Xiaoxian, 2019-08-08:
@ArcticCat yeah cartographer need accurate predict to do scan match, when you get faster, your angular prediction from odometry become inaccurate. You can try to use IMU and publish Imu message at high frequency, that maybe helped.

ArcticCat, 2019-08-09:
But cartographer uses IMU only for finding a normal to the horizontal surface in 2D mapping, doesn't it?

ArcticCat, 2019-08-09:
@Yu-Xiaoxian , but cartographer uses IMU only for finding a normal to the the horizontal surface in 2D mapping, doesn't it?

nhphuong91, 2019-08-12:
> @Yu-Xiaoxian , but cartographer uses IMU only for finding a normal to the the horizontal surface in 2D mapping, doesn't it?

Correct! That's what they said in Cartographer doc. Even in the simulation example of turtlebot3, I saw they use a node (flat_world_imu_node) which zero-ing xy components & set z to a constant number (9.8 m/s^2) to filter IMU data before feeding it to cartographer node.

Yu-Xiaoxian, 2019-08-13:
> @Yu-Xiaoxian , but cartographer uses IMU only for finding a normal to the the horizontal surface in 2D mapping, doesn't it?

Exactly, pose extrapolator use IMU to predict the pose, which will be the initial guess for ceres scan matcher. And ceres scan matcher is very sensitive to initial pose, as a result an accurate orientation is very important.

closed 2020-10-16 by MichaelGrupp