cartographer-project/cartographer_ros #1369: Does the order of points in a published point cloud matter for local SLAM?

UserUsingGit, 2019-07-29:
I assigned this issue to cartographer_ros since I am using it, but it might also be an issue for cartographer. I am testing the effect of (spatially lossy) compressions on local SLAM. The dataset I am using is KITTI (2D SLAM, using only cropped velodyne data, no IMU, one whole scan sent in one message). 
I am testing a compressed and decompressed dataset with maximum deviation of 1cm, which should have (almost) no effect on local SLAM, assuming the voxel grid filter used is 15cm. Additionally, the compression and decompression modifies the order of points. I did some tests, which resulted in the following trajectories:
[image]
So the order of points seems to be the only thing that really matters in that case (correct me if I am wrong). Unfortunately, I was not able to find the reason for that in cartographer's code (I checked  CeresScanMatcher2D::Match as well as the cost function, but did not find anything). In what part of the code could the order of the points in the point cloud matter?

Here are my configuration files:
kit_2d.lua:
https://github.com/UserUsingGit/cartographer_ros/blob/master/cartographer_ros/configuration_files/kit_2d.lua
demo_kit_2d.rviz
https://github.com/UserUsingGit/cartographer_ros/blob/master/cartographer_ros/configuration_files/demo_kit_2d.rviz
kit_2d.launch
https://github.com/UserUsingGit/cartographer_ros/blob/master/cartographer_ros/launch/kit_2d.launch
demo_kit_2d.launch
https://github.com/UserUsingGit/cartographer_ros/blob/master/cartographer_ros/launch/demo_kit_2d.launch
kit_2d.urdf
https://github.com/UserUsingGit/cartographer_ros/blob/master/cartographer_ros/urdf/kit_2d.urdf

rosbag_validate gives the following output:
cartographer_rosbag_validate -bag_filename /home/user/data/kitti_2011_10_03_drive_0027/bagfiles/kitti_0027_0_000001m.bag
E0729 12:30:20.301599 22724 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.120001 s, recommended is [0.0005, 0.005] s with no jitter.
I0729 12:30:20.301928 22724 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/kitti/oxts/imu" (frame_id: "imu_link"):
Count: 4543  Min: 0.079936  Max: 0.120001  Mean: 0.103640
[0.079936, 0.083943)	                    	Count: 1 (0.022012%)	Total: 1 (0.022012%)
[0.083943, 0.087949)	                    	Count: 0 (0.000000%)	Total: 1 (0.022012%)
[0.087949, 0.091956)	                    	Count: 5 (0.110059%)	Total: 6 (0.132071%)
[0.091956, 0.095962)	                    	Count: 0 (0.000000%)	Total: 6 (0.132071%)
[0.095962, 0.099969)	                ####	Count: 980 (21.571650%)	Total: 986 (21.703720%)
[0.099969, 0.103975)	            ########	Count: 1883 (41.448383%)	Total: 2869 (63.152103%)
[0.103975, 0.107982)	                    	Count: 63 (1.386749%)	Total: 2932 (64.538849%)
[0.107982, 0.111988)	             #######	Count: 1610 (35.439137%)	Total: 4542 (99.977989%)
[0.111988, 0.115995)	                    	Count: 0 (0.000000%)	Total: 4542 (99.977989%)
[0.115995, 0.120001]	                    	Count: 1 (0.022012%)	Total: 4543 (100.000000%)
E0729 12:30:20.302032 22724 rosbag_validate_main.cc:382] Point data (frame_id: "velo_link") has a large gap, largest is 0.104733 s, recommended is [0.0005, 0.05] s with no jitter.
I0729 12:30:20.302076 22724 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/kitti/velo/pointcloud" (frame_id: "velo_link"):
Count: 4543  Min: 0.102937  Max: 0.104733  Mean: 0.103652
[0.102937, 0.103117)	                    	Count: 4 (0.088048%)	Total: 4 (0.088048%)
[0.103117, 0.103296)	                    	Count: 30 (0.660357%)	Total: 34 (0.748404%)
[0.103296, 0.103476)	                   #	Count: 203 (4.468413%)	Total: 237 (5.216817%)
[0.103476, 0.103655)	          ##########	Count: 2225 (48.976448%)	Total: 2462 (54.193264%)
[0.103655, 0.103835)	            ########	Count: 1826 (40.193703%)	Total: 4288 (94.386971%)
[0.103835, 0.104015)	                   #	Count: 185 (4.072199%)	Total: 4473 (98.459167%)
[0.104015, 0.104194)	                    	Count: 64 (1.408761%)	Total: 4537 (99.867928%)
[0.104194, 0.104374)	                    	Count: 5 (0.110059%)	Total: 4542 (99.977989%)
[0.104374, 0.104553)	                    	Count: 0 (0.000000%)	Total: 4542 (99.977989%)
[0.104553, 0.104733]	                    	Count: 1 (0.022012%)	Total: 4543 (100.000000%)

If necessary, I will provide the bagfiles too, please let me know.
Any help/guess is appreciated.
Thank you very much in advance!

UserUsingGit, 2019-08-29:
To be sure it is the order of points that affected local SLAM, I took the original Velodyne data, randomly shuffled the points in each frame and wrote them back to the binary file. Then I converted the respective data to bagfiles. I made sure the points keep the same types and accuracy as before. The results confirm the results from above.
[image]
I was not able to find the reason for this yet.

MichaelGrupp, 2020-10-16:
Cartographer expects ordered input. I can't help with your use-case, so you might need to check yourself what happens in detail if you are interested.

closed 2020-10-16 by MichaelGrupp