cartographer-project/cartographer_ros #1370: My own bag can be used in Cartographer

carleft, 2019-07-31:
Hello everyone!
I am using a Lslidar and a imu to map an envirment. But there are some error when I run 'rosbag_validate' on my sensor data:
[https://gist.github.com/carleft/f2efa81fe45bb75fed9beb6d503c8ae9](url)

and this is my configuration files:
[https://github.com/carleft/Cartographer_Lslidar_3d_SLAM](url)

and this is my rosbag:
[https://drive.google.com/open?id=1C6HxLhwipUS5UyNK8IQH1ex7NNlZWLOi](url)

When I use my bag to map a 3d map, there also some error show that:

roslaunch cartographer_ros demo_lslidar.launch bag_filename:=/home/tb/2019-07-31-10-34-40.bag
[ WARN] [1564555939.322514615, 1564540485.778566788]: W0731 14:52:19.000000 17455 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)

So, How can I solve those issues?
Any help/guess is appreciated.
Thank you very much in advance!

aidamQin, 2019-08-09:
Please check if your laser frame_id in yourself is the same as Cartographer sub

MichaelGrupp, 2020-11-06:
You have to ensure that the topic name is the one that cartographer expects. Use `rosnode info` to check the topic names that cartographer subscribes to. You can use a topic `remap` in your launch file if the name of your points topic doesn't match.

closed 2020-11-06 by MichaelGrupp