cartographer-project/cartographer_ros #1371: Can't get live map with rplidar

Rhonins, 2019-08-08:
Hello ,
I am trying to make 2d map with 2D lidar and want to see it on my Jetson nano.

I have ubuntu 18.04 installed all ros packages etc.

I can run demo files with bag but I am not able to get live mapping with my RPlidar a2.
Also I can run rplidar's launch file and able to see working lidar.
After I launch file I am running rplidar from different wotkspace and providing scan topic.

When I run rviz:
I am getting "no map received" error

lua :
`-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "horizontal_laser_link",
  published_frame = "horizontal_laser_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.submaps.num_range_data = 35
TRAJECTORY_BUILDER_2D.min_range = 0.3
TRAJECTORY_BUILDER_2D.max_range = 8.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35
POSE_GRAPH.constraint_builder.min_score = 0.65

return options`

urdf : 
`

<robot name="3seviye">
  <material name="orange">
    <color rgba="1.0 0.5 0.2 1" />
  </material>
  <material name="gray">
    <color rgba="0.2 0.2 0.2 1" />
  </material>

  <link name="imu_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <box size="0.06 0.04 0.02" />
      </geometry>
      <material name="orange" />
    </visual>
  </link>

  <link name="horizontal_laser_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <cylinder length="0.05" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="vertical_laser_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <cylinder length="0.05" radius="0.03" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="base_link" />

  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu_link" />
    <origin xyz="0 0 0" />
  </joint>

  <joint name="horizontal_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="horizontal_laser_link" />
    <origin xyz="0.1007 0 0.0558" />
  </joint>

  <joint name="vertical_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="vertical_laser_link" />
    <origin rpy="0 -1.570796 3.141593" xyz="0.1007 0 0.1814" />
  </joint>
</robot>`

launch : 
`

<launch>
  <param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/3seviye.urdf" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename 3seviye.lua"
      output="screen">
    <remap from="scan" to="horizontal_laser_2d" />
  </node>
<node name="laser" pkg="tf"
type="static_transform_publisher" args="0.07 0 0 0 0 0 1 base_link laser 50"/>
  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
</launch>`

[rqt without tf:](https://imgur.com/zByCaBw)

[rqt with tf:](https://imgur.com/nkn4P4o)

[rqt tree:](https://imgur.com/HIW3pbz)

How can I get 2d mapping only with the RPLidar?

Thanks.

jonra1993, 2019-08-31:
Hi @Rhonins  could you help this issue. I am facing the same problem with a ydLidar X4.

jgaud, 2019-11-15:
@jonra1993 Did you figure out how to set it up?
I am running into the same issue.

Thanks.

jonra1993, 2019-11-15:
Hi @jgaud  yes I solve it for my ydlidar X4. Refer to this pull request https://github.com/EAIBOT/ydlidar/pull/15

[image]

jgaud, 2019-11-15:
@jonra1993 Thanks for your help, I will look into it!

MichaelGrupp, 2020-10-23:
@Rhonins remap is required. Cartographer expects certain topic names depending on how many lasers you use, see `rosnode info /cartographer_node`. Other than that, I can't help much with a custom setup besides the answers that were given from the other users.

closed 2020-10-23 by MichaelGrupp