cartographer-project/cartographer_ros #1373: How to  save or delete specific trajectory?

robot-mh, 2019-08-12:
After run cartographer_ros node and _write_state_, _load_state_filename_ multi times, I have 0~5 trajectory in file map.pbstream.

How can I delete trajectory 0~4, only reserve the last trajectory 5?

or how to save specific trajectory use _write_state_？

thank you guys. @SirVer @esteve @damienrg @damonkohler @skohlbr

ShihanWang, 2019-10-24:
I have the same question. I think we should change the WritePbSteam function code: 
delete old trojectory data in PoseGraph: constraints、submaps、Nodes...

ShihanWang, 2019-10-24:
this may be useful to solve it:
[https://github.com/googlecartographer/cartographer/issues/1474](url)

ShihanWang, 2019-11-05:
I changed the WritePbStream function code, and just save the specific trajectory with its trajectory_id=0, at the same time, remap all the constraints、submaps、nodes、imu_data、odometry_data. But deleting a specific trajectory is more complicated...

MichaelGrupp, 2020-10-16:
There is a pull request with a concept draft for that in the RFC repository. But it's not implemented here yet.

closed 2020-10-16 by MichaelGrupp

GrantZheng86, 2021-06-24:
Hi @ShihanWang,

Do you know where is the "WritePbStream" function located? I'm trying to do a similar thing.

Thanks!