cartographer-project/cartographer_ros #1374: Why does the amount of trajectory nodes output by cartographer_ros is smaller than the amount of scans in the bagfile?

UserUsingGit, 2019-08-12:
Hello. I am running Cartographer on the KITTI dataset and evaluating using the evo evaluation kit (https://github.com/MichaelGrupp/evo). For calculating RPE and APE, the amount of trajectory nodes in the ground truth data and in the SLAM data have to match (since the APE/RPE are calculated per trajectory node). Running online node and offline node and writing out the points, I realized that the number of points differ (for KITTI sequence_00, there are around 4500 frames and the SLAM result has around 30 trajectories less than the ground truth data. So cartographer seems not to write out a node for each scan. First I thought using the offline node, there will be no scan that is missed out since it is running through the bagfile without any time limitations but this does not seem to be the case. Is there an option/possibility to make cartographer write out a trajectory node for each scan taken? 
Thanks to everyone, every help/guess appreciated. 

Here is my rosbag validate:
E0812 13:40:14.479629 29659 rosbag_validate_main.cc:389] IMU data (frame_id: "imu_link") has a large gap, largest is 0.120001 s, recommended is [0.0005, 0.005] s with no jitter.
I0812 13:40:14.479897 29659 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/kitti/oxts/imu" (frame_id: "imu_link"):
Count: 4543  Min: 0.079936  Max: 0.120001  Mean: 0.103640
[0.079936, 0.083943)	                    	Count: 1 (0.022012%)	Total: 1 (0.022012%)
[0.083943, 0.087949)	                    	Count: 0 (0.000000%)	Total: 1 (0.022012%)
[0.087949, 0.091956)	                    	Count: 5 (0.110059%)	Total: 6 (0.132071%)
[0.091956, 0.095962)	                    	Count: 0 (0.000000%)	Total: 6 (0.132071%)
[0.095962, 0.099969)	                ####	Count: 980 (21.571650%)	Total: 986 (21.703720%)
[0.099969, 0.103975)	            ########	Count: 1883 (41.448383%)	Total: 2869 (63.152103%)
[0.103975, 0.107982)	                    	Count: 63 (1.386749%)	Total: 2932 (64.538849%)
[0.107982, 0.111988)	             #######	Count: 1610 (35.439137%)	Total: 4542 (99.977989%)
[0.111988, 0.115995)	                    	Count: 0 (0.000000%)	Total: 4542 (99.977989%)
[0.115995, 0.120001]	                    	Count: 1 (0.022012%)	Total: 4543 (100.000000%)
E0812 13:40:14.479979 29659 rosbag_validate_main.cc:382] Point data (frame_id: "velo_link") has a large gap, largest is 0.104733 s, recommended is [0.0005, 0.05] s with no jitter.
I0812 13:40:14.480007 29659 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/kitti/velo/pointcloud" (frame_id: "velo_link"):
Count: 4543  Min: 0.102937  Max: 0.104733  Mean: 0.103652
[0.102937, 0.103117)	                    	Count: 4 (0.088048%)	Total: 4 (0.088048%)
[0.103117, 0.103296)	                    	Count: 30 (0.660357%)	Total: 34 (0.748404%)
[0.103296, 0.103476)	                   #	Count: 203 (4.468413%)	Total: 237 (5.216817%)
[0.103476, 0.103655)	          ##########	Count: 2225 (48.976448%)	Total: 2462 (54.193264%)
[0.103655, 0.103835)	            ########	Count: 1826 (40.193703%)	Total: 4288 (94.386971%)
[0.103835, 0.104015)	                   #	Count: 185 (4.072199%)	Total: 4473 (98.459167%)
[0.104015, 0.104194)	                    	Count: 64 (1.408761%)	Total: 4537 (99.867928%)
[0.104194, 0.104374)	                    	Count: 5 (0.110059%)	Total: 4542 (99.977989%)
[0.104374, 0.104553)	                    	Count: 0 (0.000000%)	Total: 4542 (99.977989%)
[0.104553, 0.104733]	                    	Count: 1 (0.022012%)	Total: 4543 (100.000000%)  

Here is my cartographer fork:
https://github.com/UserUsingGit/cartographer_ros
I am using the configuration, launch and urdf files:
kit_2d.lua
demo_kit_2d.launch
kit_2d.urdf

Here is a gdrive link to the bagfile used:
https://drive.google.com/open?id=1CUFOZyvhm9L0yBDbMblxwvFy1UD8lPxk

One other small question. Is this repository of cartographer_ros still under active development/maintainance or did it switch to another one? I feel like there has not happened alot

ojura, 2019-08-12:
There is a motion filter inside cartographer which discards nodes if you are not moving much. You can disable it by setting `TRAJECTORY_BUILDER_2D.motion_filter.max_time_seconds = 0.`.

UserUsingGit, 2019-08-13:
Thank you so much for the help! Using the online node, the amount of trajectories fits now. When I run the offline node on the same lua and bag file, there are still 10 trajectory nodes missing (rosbag info outputs 4544 msgs, the amount of nodes written out is 4534). Is there any obvious thing I am overseing?

MichaelGrupp, 2019-08-13:
You might also need to tweak the `num_accumulated_range_data` parameter, see my comment here: https://github.com/googlecartographer/cartographer_ros/issues/1297#issuecomment-496217863

Regarding your initial problem, this might also help: https://github.com/MichaelGrupp/evo/issues/152#issuecomment-473660536
There are also "KITTI to rosbag" tools that merge poses and timestamps, e.g. https://github.com/ethz-asl/kitti_to_rosbag

UserUsingGit, 2019-08-14:
Thank you for the great help! I will have a look into the KITTI to rosbag tool as well as the kitti_poses_and_timestamps_to_trajectory.py script. 
The parameters num_accumulated_range_data, num_point_clouds and num_subdivisions_per_laserscan where all set properly, so this was not the problem. But I found the problem in the source file of the offline node, right before the actual RunOfflineNode() function:

// We publish tf messages one second earlier than other messages. Under
// the assumption of higher frequency tf this should ensure that tf can
// always interpolate.
const ::ros::Duration kDelay = ::ros::Duration(1.0);

As I understand it, the command starts the offline node 1 second later, while the bagfile already publishes messages. Since KITTI's sensor frequency is 10 Hz, this leads to the first 10 trajectories getting lost every run. Does that make sense? I recompiled using no delay, and the amount of trajectories is now exactly the same as in the rosbag file. 
In your opinion, can i assume that the KITTI frequency is low enough to not have problems with tf interpolation?

closed 2020-10-16 by MichaelGrupp

gaussiangit, 2021-04-15:
@MichaelGrupp I have another camera along with lidar. Number of camera images and poses/node_index should match in my case! How can I achieve that ?

marcoluis97, 2022-12-02:
Greetings @UserUsingGit 

> As I understand it, the command starts the offline node 1 second later, while the bagfile already publishes messages. Since KITTI's sensor frequency is 10 Hz, this leads to the first 10 trajectories [...]

Were you successful in getting synchronized messages after removing the 1 second delay? If so, could you guide me in order to achieve the same results?