cartographer-project/cartographer_ros #1376: assets writer map output (pmg) doesnt match submaps and doenst show all obstacles

Leisa2019, 2019-08-12:
Hey there, 
I've been having an issue with converting my cartographer pbstream-output to a usable pgm-file using the asset writer. The results of the submaps shown in RViz show all the obsticals that I need. After saving the map as .gpm, I only seem to get parts of that result. What I already tried is changing the miss/hit probabilities in the .lua-file, which only made it worst. I also tried setting insert_free_space to false, which resulted in a better obstical recognition but in a gray and black map, that includes basically everything that the sensors ever recorded and that is overlapped. 
I get that the asset writer is there for the recombination of the submaps including propabilities or color filters, but what I don't get is that I can't tune it in a way that I can receive a gpm that resembles the RVIZ-View.
 Why does the hit-probability have to be >0.5 and the miss probability <0.5? 
Does anyone else have this issue and maybe a solution or a tip?

Leisa2019, 2019-08-13:
[image]
[image]
Here are some pictures showing the issue.

facontidavide, 2019-10-10:
This is a very well known problem (duplicate of #963

I am sending a PR to fix it.

gameinskysky, 2020-01-15:
Thanks for your help

closed 2020-06-09 by wohe