cartographer-project/cartographer_ros #1380: Install on Debian error

SimonDahrs, 2019-08-27:
I'm trying to install cartographer_ros on my Raspberry Pi 4 with Debian Buster. Installing ROS goes without a problem, but when I try to install cartographer using [these instructions](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#building-installation). However it errors on this command: 
```bash
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
```
This is the output it provides:
```
cartographer_ros: No definition of [visualization_msgs] for OS [debian]
cartographer_rviz: No definition of [rviz] for OS [debian]
```

I tried multiple supposed solutions found online but non of them works. 
Does anyone know how to fix this?

Ocoras, 2019-12-02:
Receiving similar error when building on Arch:

```
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ceres-solver: No definition of [libgoogle-glog-dev] for OS [arch]
cartographer: No definition of [lua5.2-dev] for OS [arch]
cartographer_ros: Cannot locate rosdep definition for [rosunit]
cartographer_rviz: Cannot locate rosdep definition for [rviz]
cartographer_ros_msgs: Cannot locate rosdep definition for [message_runtime]
```
I am attempting using ros2, so I think the error stems from Cartographer definitions between ros1 & ros2 - if so, is there a correct way to install Cartographer for ros2?

bochen87, 2020-06-17:
The solution is in the error description. You’re missing those libraries. Either your distro has those libraries via apt or similar or you need to compile from source.

closed 2020-06-17 by bochen87