cartographer-project/cartographer_ros #1389: How to use two 2d lidars in cartographer 2d?

bill4u, 2019-09-17:
Do I need to create a node to merge two 2d lidars, if so, How ? Or Does cartographer offer some interfaces ? Any suggestion?

tulku, 2019-09-17:
when using cartographer_ros you can subscribe directly to the different lidar topics.

When creating the configuration file use the option:
```
options {
   (...)
   num_laser_scans = 2
   (...)
}
```
Cartographer will the subscribe to `scan_1` and `scan_2` topics which you can remap to the correct topics in your setup.

The reference for the different options can be found [here](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html).

bill4u, 2019-09-18:
> when using cartographer_ros you can subscribe directly to the different lidar topics.
>
> When creating the configuration file use the option:
>
> ```
> options {
>    (...)
>    num_laser_scans = 2
>    (...)
> } [...]

Great! THX a lot!

bill4u, 2019-09-18:
@tulku @esteve @SirVer I am sorry to disturb you, But I have encountered some difficulty and have no ideas. I change "num_laser_scans = 2" and publish "scan_1" and "scan_2", publish the tf. However, cartographer doesn't work as well as only one lidar.
Here's some info.
[image]

setting1:
[image]

            result:
[image]

            log:

[log1.txt](https://github.com/googlecartographer/cartographer_ros/files/3626878/log1.txt)

setting2:
[image]

            result:
[image]

            log:
[log2.txt](https://github.com/googlecartographer/cartographer_ros/files/3626881/log2.txt)

Here is my .lua file:
[cuutuu.lua.txt](https://github.com/googlecartographer/cartographer_ros/files/3626886/cuutuu.lua.txt)

My Rosbag is here: https://pan.baidu.com/s/1ijQWyv3_mmNARvDmg9PbYw  CODE: gmpu

THANKS for your advices.

bill4u, 2019-09-18:
[image]

tulku, 2019-09-18:
From your comments I understand that when using a single lidar cartographer is able to produce a better motion estimation from consecutive scans and when using 2 lidars, cartographer is failing to produce that estimation.

I noticed that you don't have an odometry source nor IMU, so cartographer has no other sensor to create the motion estimation.

I'm also working with multiple lidars and I have the same problem. However, when using an odometry source works better.

I would double check that the lidar poses are correct (if there is some overlap on the scans, that they match properly) and then keep playing with both the ceres scan matcher and the correlative scan matcher until you get a better result.

Let me know if you find values that make this work properly :)

bill4u, 2019-09-20:
> From your comments I understand that when using a single lidar cartographer is able to produce a better motion estimation from consecutive scans and when using 2 lidars, cartographer is failing to [...]

I use the IMU and get somehow better result, but still doesn't work properly. Then I merge two lidars data myself and feed into the cartographer, the result is much better. But it is strange that if the robot is static, the TF between **/map** and **/laser** may moving. I have a doubt that if two lidars need the  synchronization of timestamp? Maybe the "imu_raw" needs to be smoothed using imu_tools? Any ideas?

tulku, 2019-09-20:
Interesting observation that making one lidar message before works better. That reminded me of the setting:
```
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = N.
```
I did some quick tests with that, setting to the number of lidars in case cartographer needs to build one scan out of all, but did not have good results. You might want to give it a try.

When disabling odom and imu and using multiple lidars, the localization is moving a lot, so /map to /lidar will change because the robot pose will change, which is in between that transform. And I see the same here. And that's because cartographer's scan matcher is given very wrong results.

It might be that there are some conditions that the scans should have in order to work directly. Unfortunately I don't know them yet. We will find out though :)

closed 2019-10-17 by bill4u