cartographer-project/cartographer_ros #1393: Request for help tuning cartographer in 3d

Ellon, 2019-09-30:
Hi,

I'm trying to tune cartographer on a custom bag for a vehicle in 3d with no success. :frowning_face: My vehicle accumulates considerable drift on Z and is not able to loop close when rejoining it's trajectory. Could anyone give me some tips on how to tune the algorithm for my case ?

Here's the information requested on the boilerplate:
* [my branch of cartographer_ros](https://github.com/Ellon/cartographer_ros/tree/vehicle) with my config and launch files;
* [the bag](https://drive.google.com/file/d/1buJqgC3mQgXJ9gjMPzlfOGR2BfjTYciP/view?usp=sharing) used during my tests;
* [rosbag_validate output](https://gist.github.com/Ellon/681fbec42054c02e631b78f594da8259).

The static sensor transforms are published on `/tf` on the bag. They are the same as the ones of the URDF file that you can find on the branch.

X-ray of my results:

* XY
[image]

* XZ
[image]

* YZ
[image]

Ellon, 2019-10-09:
@gaschler @ojura Would you mind taking a look at it if you have time ?
(I'm pinging you two based on comments on related issues... :) )

eithwa, 2021-08-23:
[image]
Due to the huge map size, I use a large search window
```
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher_3d.linear_xy_search_window = 50.
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher_3d.linear_z_search_window = 3.
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher_3d.angular_search_window = math.rad(30.)
POSE_GRAPH.constraint_builder.min_score = 0.45
```