cartographer-project/cartographer_ros #1394: 3D SLAM using two 2D Lidars

anymate98, 2019-10-02:
I'm trying to do 3D SLAM using cartographer. I recently tried 2D SLAM with only one 2D Lidar, and it performed pretty well. Now I want to do a 3D SLAM, but the only sensors I can use are two 2D Lidars. Maybe I can additionally use IMU as well, but nothing more would be possible. I already read the documentation and it only tells me that to perform 3D SLAM, it needs an IMU input. So is two 2D Lidars(one for vertical scan, and the other for horizontal scan) and IMU sufficient enough to perform 3D SLAM? If not, what more do I need?

AVR-TEC, 2019-12-03:
Hello, I have the same inquiry

Can the data from the 2D backpack be used for Cartographer 3D since it has both vertical and horizontal scans in the bag file? or the scan information is not enough for the algorithm?

I have been trying to make the system work in 3D but i cant get the correct configuration of parameters in the LUA file, it doesn't seem to work correctly and always diverges or doesn't move at all from the initial position. I wonder if you already did such experiment and if it was successful could share the correct configuration file? or is it impossible to make it work with just two 2D lidars?.

Thank you.

MichaelGrupp, 2020-10-16:
We recommend using a 3D lidar for 3D mode. If your idea works would need some experimentation.

Closing as inactive.

closed 2020-10-16 by MichaelGrupp