cartographer-project/cartographer_ros #1395: Request for help building cartographer_ros: catkin_make_isolated --install --use-ninja

JiyongOh, 2019-10-02:
Hello everyone,

I'm trying to install cartographer_ros package in the environment of Ubuntu 16.04 and ROS kinetic.
I followed the instruction written in the [web page](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html). 
Unfortunately, I face the error below after performing the last command (catkin_make_isolated --install --use-ninja).
Could anyone give me some tips how to overcome the following error and install the package?

----------------------------------- < messages > --------------------------------------------------------------------- 

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/~~~~/catkin_ws/install_isolated/env.sh'
==> cmake /home/~~~~/catkin_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/~~~~/catkin_ws/install_isolated -G Ninja in '/home/~~~~/catkin_ws/build_isolated/cartographer/install'
-- Build type: Release
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   iostreams
--   regex
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
CMake Error at /home/~~~~/catkin_ws/install_isolated/lib/cmake/Ceres/CeresConfig.cmake:88 (message):
  Failed to find Ceres - Missing requested Ceres components: [SuiteSparse]
  (components requested: [SuiteSparse]).  Detected Ceres version: 1.13.0
  installed in: /home/~~~~/catkin_ws/install_isolated with components:
  [LAPACK, SchurSpecializations, OpenMP].
Call Stack (most recent call first):
  /home/~~~~/catkin_ws/install_isolated/lib/cmake/Ceres/CeresConfig.cmake:392 (ceres_report_not_found)
  CMakeLists.txt:32 (find_package)

CMake Error at CMakeLists.txt:32 (find_package):
  Found package configuration file:

    /home/~~~~/catkin_ws/install_isolated/lib/cmake/Ceres/CeresConfig.cmake

  but it set Ceres_FOUND to FALSE so package "Ceres" is considered to be NOT
  FOUND.

anymate98, 2019-10-02:
Well, it seems that SuiteSparse, a dependency of ceres-solver is missing and not found by cmake.
Maybe you can try installing SuiteSparse by using command: `sudo apt-get install libsuitesparse-dev` and retry.

JiyongOh, 2019-10-02:
Thank you for your comment.
In fact, I thought like you and tried to install SuiteSparse.
However, SuiteSparse (libsuitesparse-dev) is already installed.
Maybe Do I have to set the path (or environment variable) of libsuitesparse-dev?
If so, please let me know how to set the path.

anymate98, 2019-10-02:
Maybe a duplicate of https://github.com/googlecartographer/cartographer/issues/993?

JiyongOh, 2019-10-04:
@ anymate98
Thanks again.
I found the issue that you indicated and I followed the instructions provided in the page.
However, I could not avoid the same error message before opening this issue.

JiyongOh, 2019-10-07:
I solved this problem by reinstalling cartographer_ros package after removing and reinstalling ROS.

closed 2019-10-07 by JiyongOh