cartographer-project/cartographer_ros #1396: failure to run Deutsches Museum demo after fresh cartographer install

snusnumrick, 2019-10-04:
I was following instructions how to install cartographer. (Ubuntu 18.04 ROS Melodic)

When I try to tun demo using commands from 
https://google-cartographer-ros.readthedocs.io/en/latest/demos.html
I get following terminal output and following rozviz screen. What could be wrong?

[image]

`started roslaunch server http://josephine:39673/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [21753]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3d0f9668-e6d8-11e9-8817-00d8610ffe53
process[rosout-1]: started with pid [21764]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [21768]
process[cartographer_node-3]: started with pid [21772]
process[cartographer_occupancy_grid_node-4]: started with pid [21773]
process[rviz-5]: started with pid [21779]
process[playbag-6]: started with pid [21789]
terminate called after throwing an instance of 'boost::thread_interrupted'
[cartographer_node-3] process has died [pid 21772, exit code -11, cmd /media/anton/HDD/ros/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /media/anton/HDD/ros/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_3d.lua points2_1:=horizontal_laser_3d points2_2:=vertical_laser_3d __name:=cartographer_node __log:=/home/anton/.ros/log/3d0f9668-e6d8-11e9-8817-00d8610ffe53/cartographer_node-3.log].
log file: /home/anton/.ros/log/3d0f9668-e6d8-11e9-8817-00d8610ffe53/cartographer_node-3*.log
[cartographer_occupancy_grid_node-4] process has died [pid 21773, exit code -11, cmd /media/anton/HDD/ros/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_occupancy_grid_node -resolution 0.05 __name:=cartographer_occupancy_grid_node __log:=/home/anton/.ros/log/3d0f9668-e6d8-11e9-8817-00d8610ffe53/cartographer_occupancy_grid_node-4.log].
log file: /home/anton/.ros/log/3d0f9668-e6d8-11e9-8817-00d8610ffe53/cartographer_occupancy_grid_node-4*.log
`

fawadahm, 2019-10-08:
Any updates on this? I am facing the same issue

helloparanoidandroid, 2020-02-13:
@snusnumrick Hi, met the same issue with demo-bag on Ubuntu 16.04 and ROS Kinetic. Everything worked fine until I tried to update the codes from version 1.0.0 to the fresh version on github. After that, cartographer never worked with either my own bag or the demo-bags again, even if I reinstalled the version 1.0.0. Error log looked the same with yours . Have you solved the problem? Thanks.