cartographer-project/cartographer_ros #1401: Cartographer in very big environments

francescoricciuti, 2019-10-23:
Dear community,

I'm writing to ask a general question regarding the performances of Cartographer in very big environments.

I'm planning to map a very big environment ( around 1 square km) and I'm trying to understand which is the best way to approach this problem.

From your own experience, is it possible to map such a big environment in a single map (a single .pbstream file) and, if it is, what would be the performances of pure localization on that map?

Consider that I'll map with a relatively slow robot, so I expect the mapping process to last very long.

Is there a way to "split" the map in different smaller maps (whatever it means to "split" a map) and change them at runtime?

Thanks in advance for your help

Best,
Francesco

AlexReimann, 2019-10-23:
2D or 3D?

francescoricciuti, 2019-10-23:
3D - I'm working with a Robosense RS-16

AlexReimann, 2019-10-23:
Should be doable, probably the map is quite large and takes a lot of RAM but nothing crazy if you tune the parameters properly.
Initial localization might take very long for pure localization.

How slow is relatively slow?

Map splitting should be also possible, there are services to load pbstreams on the fly. Never did that though.

francescoricciuti, 2019-10-24:
> How slow is relatively slow?

Relatively slow is around 1 m/s.

> Map splitting should be also possible, there are services to load pbstreams on the fly. Never did that though.

Do you happen to have any reference I could look at? I've looked for these services (not very accurately to be honest) and I couldn't find anything.

My worry is that if I load a different .pbstream I need to relocalize on it and then I would need to write an autonomous routine for this purpose (which I would prefer not to do)

AlexReimann, 2019-10-25:
By default cartographer solves the `kidnapped robot` problem when starting.

So if you load a new map I would guess it tries to figure out itself where it is in the new map. Can take a while though and you have to figure out if it localized (correctly) yourself.

Yeah, it seems like there is no ROS interface for that. Pretty sure I saw this somewhere in the code base, though.

AmineDh98, 2024-03-28:
> By default cartographer solves the `kidnapped robot` problem when starting.
> 
> So if you load a new map I would guess it tries to figure out itself where it is in the new map. Can take a while though and you have to figure out if it localized (correctly) yourself.

I am also trying to do pure localization in a huge map and having the problem of localizing after a very long time (in pure localization mode). 
Do you know a way to optimize this and reduce the localization time in big maps ? (what parameters to tune ?)
Can I somehow choose/limit  the search area where it searches for global constraints ? For example the user knows where approximately the robot is and this information can help cartographer to localize easier.