cartographer-project/cartographer_ros #1403: pure_localization_trimmer is on but keeps building the map and no localization

artemiialessandrini, 2019-10-26:
Pure localization behaviour is not correct.
Although pure_localization_trimmer is on, cartographer keeps rebuilding the map.
[Video attachment](https://yadi.sk/i/3d2dXdRylKqBvw) and Image
[image]

localization .lua params are:
`TRAJECTORY_BUILDER.pure_localization_trimmer = {
  max_submaps_to_keep = 3,
}
POSE_GRAPH.optimize_every_n_nodes = 20`

[launch file](https://gist.github.com/artemiialessandrini/2988d729a05213dcdf399f5cc02f0f6c) is a bit tricky, I'm comparing cartographer trajectory with ground truth, so having two different ns.
This might be an issue.

Also attaching [.bag file](https://yadi.sk/d/IpikQbdg3hmzfw) and [urdf_launch_pbstream.zip](https://github.com/googlecartographer/cartographer_ros/files/3774741/urdf_launch_pbstream.zip)

Any suggestions, tips would be appreciated, also feel free to ask any additional info.
Thanks to Cartographer and ROS community!

artemiialessandrini, 2019-10-30:
Similar #1015

artemiialessandrini, 2019-10-30:
cartographer_ros and cartographer aren't outdated
I'm using:
  ``` 
TRAJECTORY_BUILDER.pure_localization_trimmer = {
  max_submaps_to_keep = 3,
}
```

But if uncommenting trigger straightly from _cartographer/configuration_files/trajectory_builder.lua_, pure localization works.
I'm having two .lua files, when [wood_car_2d_localization.lua](https://gist.github.com/artemiialessandrini/dc01375fc2d59d334e353e96088c52e1) includes [wood_car.lua](https://gist.github.com/artemiialessandrini/884b733ce0a2841a6515077d2e412f18) with _options_, maybe this could be the reason. (conflict?)

aq44634, 2019-11-30:
Me too,so sad.

Maibaf, 2020-03-03:
I am using *ros2 dashing*, *cartographer* from branch *ros2* with commit https://github.com/googlecartographer/cartographer/commit/3b513dfe68dd25df966a841c65058a5f4c1301cf and *cartographer_ros* from branch *dashing* with commit 001f3970fde9c2080f2ba9a0ef76950ec3fbef93.

The option `TRAJECTORY_BUILDER.pure_localization_trimmer` does not exists for me.
Setting `TRAJECTORY_BUILDER.pure_localization = true` results in the already mentioned behavior, that cartographer still creates a map, so overriding the loaded one.

Does anyone has some clues why the pure localization mode does not work as expected. Or do i misunderstand the mode?

artemiialessandrini, 2020-04-13:
@Maibaf doesn't work for me this way... Are you loading the .pbstream map frozen?

artemiialessandrini, 2020-04-14:
Fixed by decerasing ```POSE_GRAPH.constraint_builder.sampling_ratio``` and 
increasing ```POSE_GRAPH.global_sampling_ratio```, based on [1258 comment](https://github.com/cartographer-project/cartographer_ros/issues/1258#issuecomment-481717247), closing.

closed 2020-04-14 by artemiialessandrini

WilliamWoo45, 2020-04-25:
@artemiialessandrini

WilliamWoo45, 2020-04-25:
@artemiialessandrini 
Hi!
I'm also playing with the pure localization mode of Cartographer. However, I'm wondering the **exact working principle** of the pure localization, where three possible explanations (I assumed) are listed below for your reference:

1. In the [Cartographer ROS Localization-only documentation](https://google-cartographer-ros.readthedocs.io/en/latest/going_further.html#localization-only), it says that "the localization-only mode of Cartographer which will run SLAM against the existing map and won’t build a new one". Does this mean that when runing the pure localization mode, actually the Cartographer is running a **new SLAM process without updating the map**?

2. When running the pure localization mode, is the **submap** matched with the **pre-built overall map** of the environment? If so, then maybe parameters like `mapping_constraints_constraint_builder_{2d, 3d}_constraints` and `mapping_constraints_constraint_builder_{2d, 3d}_scores` are used to judge whether the pure localization is successful? Details you may refer to this [1258 comment](https://github.com/cartographer-project/cartographer_ros/issues/1258#issuecomment-481612460)

3. Do you have any idea what parameter is the `global_localization_min_score`? The [Cartographer configuration file](https://github.com/cartographer-project/cartographer/blob/2bd987ffb40e2d6667682c5c96432173f371094f/docs/source/configuration.rst) says it is the "Threshold below which global localizations are not trusted". However, may I understand this as a threshold to measure whether the pure localization is successful or not? An example value is set at `global_localization_min_score = 0.6` in [pose_graph.lua](https://github.com/cartographer-project/cartographer/blob/bcd5486025df4f601c3977c44a5e00e9c80b4975/configuration_files/pose_graph.lua#L21)

What is your opinion? Any hint or advice is appreciated.
Thanks!

Best Regards,
William

zhaozhouyang, 2025-02-19:
In pure localization mode, unlike in mapping mode, the cartographer_occupancy_grid_node cannot be started. I encountered this issue, but it worked fine after disabling it. I hope this helps others.