cartographer-project/cartographer_ros #1404: Regarding timestamp of a pose in trajectory in Cartographer

vanshilshah97, 2019-10-27:
How to access the timestamp of each pose from a .pbstream file ?
I am using the following code to access the rotation and translation of a pose.

#include "cartographer/io/proto_stream.h"
#include "cartographer/io/proto_stream_deserializer.h"
#include "cartographer/mapping/pose_graph.h"
#include "cartographer_ros/msg_conversion.h"
#include "cartographer_ros/time_conversion.h"
#include "cartographer_ros/split_string.h"
#include "geometry_msgs/TransformStamped.h"
#include <iostream>
#include <fstream>

namespace cartographer_ros 

{

namespace carto = ::cartographer;

void ExportPbstream(const std::string& pbstream_filename) 
{
  carto::io::ProtoStreamReader reader(pbstream_filename);
  carto::io::ProtoStreamDeserializer deserializer(&reader);

  carto::mapping::proto::PoseGraph pose_graph_proto = deserializer.pose_graph();
  
    for (size_t trajectory_id = 0; trajectory_id < pose_graph_proto.trajectory().size();
       ++trajectory_id) 
  {
    const carto::mapping::proto::Trajectory& trajectory_proto =
        pose_graph_proto.trajectory(trajectory_id);

        std::ofstream myfile;
		myfile.open("trajectory.csv");
		myfile<<"x,y,z,q_x,q_y,q_z,q_w \n";
    

    for (int i = 0; i < trajectory_proto.node_size(); ++i) 
    {
      const auto& node = trajectory_proto.node(i);
    // node.pose() contains the pose...
      
      double x= node.pose().translation().x();
      double y= node.pose().translation().y();
      double z= node.pose().translation().z();
      double q_x = node.pose().rotation().x();
      double q_y = node.pose().rotation().y();
      double q_z = node.pose().rotation().z();
      double q_w = node.pose().rotation().w();

      
      myfile<<x<<","<<y<<","<<z<<","<<q_x<<","<<q_y<<","<<q_z<<","<<q_w;
      myfile<<"\n";

      
    }

  
  myfile.close();

  }

}

}

int main(int argc, char *argv[])

{

	cartographer_ros::ExportPbstream("/home/vanshil/Downloads/b3-2016-04-05-14-14-00.bag.pbstream");
	std::cout << "succesfully written ab paper publish hoga " <<std::endl;
	return 0;
}

artemiialessandrini, 2019-10-28:
please, bring up question to the proper formatting

artemiialessandrini, 2019-10-28:
Related to #856 #332

artemiialessandrini, 2019-10-28:
Also kindly check https://github.com/googlecartographer/cartographer_ros/issues/856#issuecomment-476224227, to get PoseGraph as dict

anika001, 2020-05-06:
Hi @vanshilshah97 . I am facing the same problem as yours described above for getting the timestamp of the trajectory  poses  on cartographer . Have you manged to get it ? if yes I would appreciate if you can give me more details and explain how you did it . Thank you in advance.

vanshilshah97, 2020-05-06:
for (int i = 0; i < trajectory_proto.node_size(); ++i) 
    {
      const auto& node = trajectory_proto.node(i);
    // node.pose() contains the pose...
      
      **float timestamp=(node.timestamp()-621355968000000000.0)/10000000.0;**
      double x= node.pose().translation().x();
      double y= node.pose().translation().y();
      double z= node.pose().translation().z();
      double q_x = node.pose().rotation().x();
      double q_y = node.pose().rotation().y();
      double q_z = node.pose().rotation().z();
      double q_w = node.pose().rotation().w();

      
      std::cout<<(node.timestamp()-621355968000000000.0)/10000000.0;
      std::cout<<"\n";

      myfile<<timestamp<<","<<x<<","<<y<<","<<z<<","<<q_x<<","<<q_y<<","<<q_z<<","<<q_w;
      myfile<<"\n";

      
    }

I did it like this . I had worked on it a lot before so i don't remember the full details, but the subtraction converts the timestamp to normal milliseconds. it's some thing regarding the timestamp format

closed 2020-05-06 by vanshilshah97

anika001, 2020-05-06:
@vanshilshah97  thank you for your fast response. I have another question if you do not mind. I am working on cartographer and KITTI datasets and i want to perform trajectory evaluation and i have a problem with the timestamp format which is different: the timestamp i get from cartographer is a ros time with something like 1.31738e+09 and the timestamp in kitti is like this 3.006768e+00 , how can I evaluate in this case ? cause for evaluation, we need to do timestamp matching to compute the errors in positions and if the timestamps are not in the same format I get huge errors in the positions . Do you have any idea ? Thank you again.

vanshilshah97, 2020-05-06:
@anika001 I also had a same issue while working with cartographer and kiiti .
I would have sent you all my codes but right now they are in the lab's PC which is inaccessible due to lockdown in India.

What you can do to solve this is that take the first timestamp as reference and then subsequently subtract each new timestamp from the first one. DO this for both kitti and cartographer. Then they would align.

anika001, 2020-05-06:
@vanshilshah97  Thank you for your answer. I was thinking indeed to substract the first timestamp but I was not sure if it is a kind of cheating in a way that i modified the algorithm somehow. 
Thank you for your response again. I would appreciate if you can share your codes after the lockdown.