cartographer-project/cartographer_ros #1405: Cartographer rosbag validate segmentation fault.

TheCynosure, 2019-10-28:
I am running cartographer_rosbag_validate with the following command:
```rosrun cartographer_ros cartographer_rosbag_validate -bag_filename 2014-10-07-12-58-30.bag```

The output is:
```Segmentation fault (core dumped)```

[Here is the bag file that causes the segmentation fault.](https://drive.google.com/file/d/1i7RlzAbIoVkKpZGa7TcJaO3kzSf7KI3D/view?usp=sharing)

Currently running on Ubuntu 18.04, gdb shows it is something with the RangeDataChecker, maybe a bad message?

MichaelGrupp, 2020-10-16:
This could be a bug, it shouldn't crash. Could you identify what causes this crash?

RBarron01, 2020-11-25:
Has there been any resolution to this? I'm getting the same error. I compiled with debugging symbols and ran in GDB: 

`(gdb) r -bag_filename="2020-11-18-21-28-20.bag"` output is

Starting program: /home/lidar/catkin_ws/install_isolated/bin/cartographer_rosbag_validate -bag_filename="2020-11-18-21-28-20.bag"
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
W1125 07:10:29.883761 79117 rosbag_validate_main.cc:166] Sensor with frame_id "velodyne" measurements overlap in time. Previous range message, ending at time stamp 637413533055620413, must finish before current range message, which ranges from 637413533055619993 to 637413533055626421
W1125 07:10:29.884120 79117 rosbag_validate_main.cc:166] Sensor with frame_id "velodyne" measurements overlap in time. Previous range message, ending at time stamp 637413533055674026, must finish before current range message, which ranges from 637413533055673895 to 637413533055680323
W1125 07:10:29.884275 79117 rosbag_validate_main.cc:166] Sensor with frame_id "velodyne" measurements overlap in time. Previous range message, ending at time stamp 637413533055846184, must finish before current range message, which ranges from 637413533055846093 to 637413533055852521

Program received signal SIGSEGV, Segmentation fault.
0x000055555557a6d7 in cartographer_ros::(anonymous namespace)::Run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()

 `(gdb) bt` output is
#0  0x000055555557a6d7 in cartographer_ros::(anonymous namespace)::Run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
#1  0x0000555555578dd7 in main ()

I edited the `rosbag_validate_main.cc` file to print the message numbers where it fails. The error occurs at `range_data_checker.CheckMessage(*msg);` which I believe is line 272 in the original file.

Our github repository is https://github.com/JohannesByle/ingenium_cartographer.git

RBarron01, 2020-12-02:
@MichaelGrupp I added an if statement in 'rosbag_validate_main.cc' to catch the empty frames. Seems pretty hackish so I'm not forking it yet. Not sure if this is indicative of deeper issues in our data collection process. Any advice is welcome!

`221     if (!point_cloud.empty()){
222       *to = std::get<1>(point_cloud_time);
223       *from = *to + cartographer::common::FromSeconds(point_cloud[0].time);
224       Eigen::Vector4f points_sum = Eigen::Vector4f::Zero();
225       for (const auto& point : point_cloud) {
226         points_sum.head<3>() += point.position;
227         points_sum[3] += point.time;
228       }
229       if (point_cloud.size() > 0) {
230         double first_point_relative_time = point_cloud[0].time;
231         double last_point_relative_time = (*point_cloud.rbegin()).time;
232         if (first_point_relative_time > 0) {
233           LOG_FIRST_N(ERROR, 1)
234               << "Sensor with frame_id \"" << message.header.frame_id
235               << "\" has range data that has positive relative time "
236               << first_point_relative_time << " s, must negative or zero.";
237         }
238         if (last_point_relative_time != 0) {
239           LOG_FIRST_N(INFO, 1)
240               << "Sensor with frame_id \"" << message.header.frame_id
241               << "\" has range data whose last point has relative time "
242               << last_point_relative_time << " s, should be zero.";
243         }
244       }
245       *range_checksum = {point_cloud.size(), points_sum};
246     }
247     else{LOG(ERROR)<<"Empty Frame...skipping.";}
`