cartographer-project/cartographer_ros #1408: I cant make mapping from Cartographer

aminballoon, 2019-10-31:
I don't have how to solve this error 
https://drive.google.com/open?id=1uH-Ih0qr4md0R9Ns_FFysMqu5_YuA3dx
this link have my .lua .launch and rqt_graph

nerellasureshkumar7, 2019-10-31:
Hi,
   Please check your frame_id of the laser_scan. By default link(laser sensor) name and frame_id of the sensor should be same. Try to change it and test. It will be more useful if you post the bag file.

aminballoon, 2019-10-31:
> Hi,
> Please check your frame_id of the laser_scan. By default link(laser sensor) name and frame_id of the sensor should be same. Try to change it and test. It will be more useful if you post the bag file.

Sorry i dont used .bag file my station have only one rplidar

nerellasureshkumar7, 2019-10-31:
> > Hi,
> > Please check your frame_id of the laser_scan. By default link(laser sensor) name and frame_id of the sensor should be same. Try to change it and test. It will be more useful if you post the bag file.
> 
> Sorry i dont used .bag file my station have only one rplidar

OK. Change the frame_id and test

aminballoon, 2019-10-31:
> OK. Change the frame_id and test

Can you tell me how to Chage name frame_id  I'm beginner for ros sorry

closed 2020-10-16 by MichaelGrupp