cartographer-project/cartographer_ros #1413: Track global pose of points on map

barzelshir, 2019-11-04:
What is the best way to track a global pose of a locations on map?
 Assuming that due to global optimizations the global pose of these locations changes over time.

blablebli-robots, 2019-11-19:
I do not have an answer, but I am facing a similar situation. My idea is as follows: first thing to do is to publish the cartographer's global pose, and somehow pair it with the current trajectory, store pairs (pose,traj). Then, whenever we need to map the stored poses we request the current local to global transform and use it to map the stored positions to the current location. I need to research more into this as at this point I do not know if the transforms are cumulative or not. Any thoughts on this?

barzelshir, 2019-11-24:
> I do not have an answer, but I am facing a similar situation. My idea is as follows: first thing to do is to extract the cartographer's local pose, and somehow pair it with the current trajectory, store pairs (pose,traj). Then, whenever we need to map the stored poses we request the current local to global transform and use it to map the stored positions to the current location. I need to research more into this as at this point I do not know if the transforms are cumulative or not. Any thoughts on this?

@blablebli-robots 

After some research I found out the way that they track global pose of landmarks.
They are doing it by saving the time of the observation and then interpolating between the two closest(in time) trajectory nodes, the only problem I have is that due to **trimming** the observation become older then the first trajectory node and it stopped tracking the landmark global pose.
Please view this related issue I posted:
https://github.com/googlecartographer/cartographer_ros/issues/1425

blablebli-robots, 2019-11-25:
Great, I see. So it should be possible then to maintain a parallel list of, in your case landmarks in mine a sub-trajectory, that it is not trimmed at any point, transforming only poses overtime using the new transforms. As if embedding the list/traj in the current landmarks/trajectory.

barzelshir, 2019-11-26:
> Great, I see. So it should be possible then to maintain a parallel list of, in your case landmarks in mine a sub-trajectory, that it is not trimmed at any point, transforming only poses overtime using the new transforms. As if embedding the list/traj in the current landmarks/trajectory. Of course, making sure it does not participate in any optimization at all.

I have a lot of global points which I need to track, avoid trimming them will cause high memory that I'm trying to avoid with the trimmer

blablebli-robots, 2019-11-26:
I hear you, my case seems to be simpler then, but somehow related to yours. I need to track sub-trajectories over time, but need to make sure that all my selected points survive over time. So I will start by creating a "shadow" of trajectory_nodes_ in pose_graph.* to store my sub-trajectory. I will then transform this sub-trajectory in RunOptimization() after the Solve call, before the LandMark transforms. I still need to figure out how lots of other stuff, but I think this is where all happens as far as trajectory/landmark transforms over time.

What do you think?

barzelshir, 2019-11-28:
> Great, I see. So it should be possible then to maintain a parallel list of, in your case landmarks in mine a sub-trajectory, that it is not trimmed at any point, transforming only poses overtime [...]

By sub-trajectory you mean multiple trajectory nodes?

blablebli-robots, 2019-12-02:
I'd like to store robot's poses while moving from point A to point B during mapping and track poses from point A to point B over time, specially after optimizations and loop closures. So by sub-trajectory I mean the segment of the global trajectory corresponding to poses between A and B inclusive. I hope I have explained myself this time.

Cheers!

barzelshir, 2019-12-03:
> I'd like to store robot's poses while moving from point A to point B during mapping and track poses from point A to point B over time, specially after optimizations and loop closures. So by [...]

If I understood correctly you are talking about multiple trajectory nodes, you can just save the time you got to point A, and get the closest trajectory node(by time), lets call it nodeA. then get the time you got to point B and get closest trajectory node, lets call it nodeB. From now on you can track the trajectory nodes id's between nodeA and nodeB. If you don't have trimming involved I think it's no problem.

blablebli-robots, 2019-12-03:
Yes you got it right. The main issue I am facing is precisely trimming, making it hard to even interpolate poses. I tried doing it externally but there is no way to deal with trimming as the metric properties from point A to point B change over time, specially in areas where mapping is trickiest. I am now trying to do it internally by discovering how the process works.

closed 2020-10-16 by MichaelGrupp