cartographer-project/cartographer_ros #1420: cartographer_ros and Intel Research Lab dataset leads to strange behaviour

sterngerlach, 2019-11-11:
Hello, I need to use Intel Research Lab dataset (available at [here](http://www2.informatik.uni-freiburg.de/~stachnis/datasets/datasets/intel-lab/intel.log.gz)) in cartographer_ros package (ROS Melodic). The dataset is in carmen-old log format, therefore it is converted to rosbag file using carmen_publisher node (see [here](https://github.com/artivis/carmen_publisher)). When I run cartographer_ros with generated bagfile, however, trajectory goes everywhere and fails to generate map. I tried every possible things I can come up with to fix this. I guess this is related to wrong tf configurations, but I cannot figure out what to do.

[image]

Looking at tf tree and node graph, everything seems fine, but actually it is not.
[image]

1. results for `cartographer_rosbag_validate`
I ran `carmen_publisher` node for generating bagfile:
```
$ rosrun carmen_publisher carmen2rosbag.py -i intel.log -o intel.bag
[INFO] [1573457171.581068, 0.000000]: Reading data from : intel.log
[INFO] [1573457171.582209, 0.000000]: Writing rosbag to : intel.bag
[INFO] [1573457171.583196, 0.000000]: Start convertion to rosbag
[INFO] [1573457171.584270, 0.000000]: Param  'robot_frontlaser_offset'  has value : 0.0
[INFO] [1573457171.585185, 0.000000]: Param  'robot_rearlaser_offset'  has value : 0.0
[INFO] [1573457177.348353, 0.000000]: Job's done.
```
```
$ rosbag info intel.bag
path:        intel.bag
version:     2.0
duration:    44:51s (2691s)
start:       Dec 06 2000 06:47:37.34 (976052857.34)
end:         Dec 06 2000 07:32:28.63 (976055548.63)
size:        34.6 MB
messages:    67461
compression: none [46/46 chunks]
types:       nav_msgs/Odometry     [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/LaserScan [90c7ef2dc6895d81024acba2ac42f369]
             tf2_msgs/TFMessage    [94810edda583a504dfda3829e70d7eec]
topics:      /FLASER   13631 msgs    : sensor_msgs/LaserScan
             /ODOM     26915 msgs    : nav_msgs/Odometry    
             /tf       26915 msgs    : tf2_msgs/TFMessage
```
then I ran `cartographer_rosbag_validate` command and got the result as shown below:
```
$ cartographer_rosbag_validate -bag_filename intel.bag
E1111 16:28:00.829424 28509 rosbag_validate_main.cc:382] Point data (frame_id: "base_link") has a large gap, largest is 3.00266 s, recommended is [0.0005, 0.05] s with no jitter.
I1111 16:28:00.829820 28509 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/FLASER" (frame_id: "base_link"):
Count: 13630  Min: 0.000006  Max: 3.002663  Mean: 0.197456
[0.000006, 0.300272)	     ###############	Count: 9988 (73.279533%)	Total: 9988 (73.279533%)
[0.300272, 0.600537)	                ####	Count: 2998 (21.995598%)	Total: 12986 (95.275131%)
[0.600537, 0.900803)	                   #	Count: 388 (2.846662%)	Total: 13374 (98.121788%)
[0.900803, 1.201069)	                    	Count: 74 (0.542920%)	Total: 13448 (98.664711%)
[1.201069, 1.501335)	                    	Count: 53 (0.388848%)	Total: 13501 (99.053558%)
[1.501335, 1.801600)	                    	Count: 47 (0.344828%)	Total: 13548 (99.398384%)
[1.801600, 2.101866)	                    	Count: 34 (0.249450%)	Total: 13582 (99.647835%)
[2.101866, 2.402132)	                    	Count: 38 (0.278797%)	Total: 13620 (99.926636%)
[2.402132, 2.702398)	                    	Count: 8 (0.058694%)	Total: 13628 (99.985329%)
[2.702398, 3.002663]	                    	Count: 2 (0.014674%)	Total: 13630 (100.000000%)
I1111 16:28:00.830135 28509 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/ODOM" (frame_id: "odom"):
Count: 26914  Min: 0.000034  Max: 3.107075  Mean: 0.099989
[0.000034, 0.310738)	   #################	Count: 23215 (86.256226%)	Total: 23215 (86.256226%)
[0.310738, 0.621442)	                  ##	Count: 2910 (10.812217%)	Total: 26125 (97.068443%)
[0.621442, 0.932146)	                    	Count: 488 (1.813183%)	Total: 26613 (98.881622%)
[0.932146, 1.242850)	                    	Count: 110 (0.408709%)	Total: 26723 (99.290329%)
[1.242850, 1.553554)	                    	Count: 57 (0.211786%)	Total: 26780 (99.502121%)
[1.553554, 1.864259)	                    	Count: 48 (0.178346%)	Total: 26828 (99.680466%)
[1.864259, 2.174963)	                    	Count: 41 (0.152337%)	Total: 26869 (99.832802%)
[2.174963, 2.485667)	                    	Count: 36 (0.133759%)	Total: 26905 (99.966560%)
[2.485667, 2.796371)	                    	Count: 7 (0.026009%)	Total: 26912 (99.992569%)
[2.796371, 3.107075]	                    	Count: 2 (0.007431%)	Total: 26914 (100.000000%)
```

2. Intel research lab dataset
I uploaded several files for reproducing the problem to GitHub Gist: [https://gist.github.com/sterngerlach/efc7de1f5cc1da16f2feeb1562aaa1e0](https://gist.github.com/sterngerlach/efc7de1f5cc1da16f2feeb1562aaa1e0)

3. Carmen log file and generated bagfile
I uploaded original log file and generated bagfile to Google Drive: [https://drive.google.com/drive/folders/1R1HKzrAi4IEtjRA8lAuRv3wnLK751R-3?usp=sharing](https://drive.google.com/drive/folders/1R1HKzrAi4IEtjRA8lAuRv3wnLK751R-3?usp=sharing)

In fact, this bagfile works great with slam_gmapping package. I also uploaded files necessary for running slam_gmapping to GitHub Gist: [https://gist.github.com/sterngerlach/b8aac3c6d6581537cd248471af3c5e1f](https://gist.github.com/sterngerlach/b8aac3c6d6581537cd248471af3c5e1f)

[image]

[image]

I hope someone could help me to fix this problem. Thank you.

Sumyer, 2020-01-08:
@sterngerlach 
I  meet the same problem with you.
What's more, in my case the topic "tf" in the rosbag converted by Intel.log  does not  show laser link.

Bitfroest, 2020-02-10:
https://github.com/googlecartographer/cartographer_ros/issues/1197

eithwa, 2021-08-21:
[image]
I refer to the setting of FabianSchurig and use a high weight to lock the drift of tf, but I don’t think this is the right approach. 
There were even four times of huge tf reversal that it could not be controlled, so I reset the trajectory.
I still don’t understand why the laser data causes such a huge jump in tf .
https://gist.github.com/eithwa/099d119137c2a4d982ec82c2e3d660a9

shrine1112, 2021-09-16:
The odometry error is too large, so mapping failed. The odometry must be corrected to work properly.
[image]

shrine1112, 2021-09-17:
Turn off laser scan motion undistortion
[image]

shrine1112, 2021-09-17:
> > The odometry error is too large, so mapping failed. The odometry must be corrected to work properly.
> > [image]
> 
> May I ask your parameter configuration?
> I tried not to use odom but it did not help to construct the map

I modified the source code of cartograher, turn off laser scan motion undistortion, and make sure scan and odom are synchronized.

shrine1112, 2021-09-17:
> > The odometry error is too large, so mapping failed. The odometry must be corrected to work properly.
> > [image]
> 
> May I ask your parameter configuration?
> I tried not to use odom but it did not help to construct the map

The map cannot be created correctly by modifying the parameters.

eithwa, 2021-09-28:
> Turn off laser scan motion undistortion
> [image]

The map you built is very clear!
Can you share your code, I want to know what changes you made.
Thank you very much!

WANGYINGYU, 2022-07-19:
> I modified the source code of cartograher, turn off laser scan motion undistortion, and make sure scan and odom are synchronized.

The map you built is very good, could you please share the intel lab dataset in rosbag format and configuration files you used? Thank you.

eithwa, 2024-12-11:
Due to the timestamps of odometry being insufficiently accurate, there is significant error when using velocity to predict positions.
After modification, using only the odometry delta for position prediction yields much better results.
[configure](https://gist.github.com/eithwa/cae1e33fd1211cac3dc90a56014a1834)

https://github.com/cartographer-project/cartographer/blob/877157a0d91788a7700221d87232d412cb3c1ef4/cartographer/mapping/pose_extrapolator.cc#L134
``` c++
transform::Rigid3d PoseExtrapolator::ExtrapolatePose(const common::Time time) {
  const TimedPose& newest_timed_pose = timed_pose_queue_.back();
  CHECK_GE(time, newest_timed_pose.time);
  if(odometry_data_.size()<2)
  {
      if (cached_extrapolated_pose_.time != time) {
    const Eigen::Vector3d translation =
        ExtrapolateTranslation(time) + newest_timed_pose.pose.translation();
    const Eigen::Quaterniond rotation =
        newest_timed_pose.pose.rotation() *
        ExtrapolateRotation(time, extrapolation_imu_tracker_.get());
    cached_extrapolated_pose_ =
        TimedPose{time, transform::Rigid3d{translation, rotation}};
  }
  return cached_extrapolated_pose_.pose;
  }

  const sensor::OdometryData& odometry_data_oldest = odometry_data_.front();
  const sensor::OdometryData& odometry_data_newest = odometry_data_.back();

  // Calculate relative position pose from oldest to newest
  const transform::Rigid3d odometry_pose_delta =
      odometry_data_oldest.pose.inverse() * odometry_data_newest.pose;

  // Update newest_timed_pose.pose with delta
  const transform::Rigid3d new_pose =
      newest_timed_pose.pose * odometry_pose_delta;

  // Calculate correction amount
  const transform::Rigid3d odometry_correction_ =
      newest_timed_pose.pose.inverse() * new_pose;

  // Return the corrected pose
  return newest_timed_pose.pose * odometry_correction_;
}
```
[image]