cartographer-project/cartographer_ros #1421: Landmark data integration without unique IDs

reinzor, 2019-11-11:
I read about landmark data integration into google cartographer: https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#static-landmarks

If landmark data requires an unique identifier, we need data association if we cannot extract the unique identifier from the landmark measurement (e.g. measured reflector). What would be the way to go to integrate reflector measurements into Google Cartographer? Should there be a separate node (outside of cartographer) that handles the data association problems (in map frame) and publishes unique landmarks? Or can this be embedded into Google Cartographer? Any suggestions?

ojura, 2019-11-11:
If you cout set up a separate node would output associations, I think it would be your best bet without having to get too much involved with the Cartographer code base.

reinzor, 2019-11-12:
Thanks for your reply. Could you perhaps also address the other option of integrating it into the Cartographer code base. Where should this be addressed in your opinion and do you think such a contribution is desired?

reinzor, 2019-11-19:
Any thoughts?

ojura, 2019-11-19:
Sorry, haven't seen your response. Upstream activity has been a bit quiet recently. But feel free to drop an rfc at googlecartographer/rfcs with a proposal to hopefully get some discussion going. 

Unfortunately, I can't help you with the exact design, as that would be quite time consuming. The Cartographer codebase is quite readable and self-explanatory, though; you just have to put some effort into understanding it and perhaps you'll come up with an idea. To give you a starting hint, I think this should go into the part with deals with global SLAM, ie. the pose graph optimizer and the constraint builder, which act as a correcting layer for local SLAM (which is the local trajectory builder, ie. laser odometry scan matcher which creates submaps). Perhaps you could implement an `AssociationFinder` or something like that which would be tasked with looking for associations and returning the appropriate constraints to `PoseGraph`.

reinzor, 2019-11-19:
Thanks for your response. I will discuss this internally.

ojura, 2019-11-19:
Looking forward to your rfc :-)  I think you can rely on Cartographer's constraint builder for closing loops and solving the kidnapped robot problem, and use a simple "nearest within search window" approach for creating associations. If you want to use just the markers for solving the kidnapped robot problem, we had some good experience with the method from [this](https://ieeexplore.ieee.org/document/5979759) paper.

maxbader, 2020-02-17:
Hi,
Is there any progress on this issue. Does the cartographer now support landmarks without unique IDs?
Are markers without orientation now supported, e.g. reflectors.

Huyhust13, 2021-07-23:
> Thanks for your response. I will discuss this internally.

Hi,
Any update for that? I'm looking forward to your solution.