cartographer-project/cartographer_ros #1422: Cartographer do localisation with a pgm map

JinzeDing, 2019-11-13:
I am a fresher in cartographer. In my project, I wanna use the cartographer instead of amcl to complete the localization task in the navigation. I have the accurate map in the pgm format. How could I load the pgm map in the cartographer to do the localization?  Or, is there any function like the pb_stream to ros_map that  could directly convert rosmap (pgm+yaml) to pbstream?

Tutorgaming, 2020-01-24:
I think you need to have a look at Cartographer's Algorithm first. It is the Graph-based Localization with the sub-map inside. If you compare the model of both "amcl" which is occupancy grid localization and "cartographer" You will found that only the image is not enough to do the localization. It must have the pose graph, sub-maps and many things to be prepared first. :)

barzelshir, 2020-03-10:
> I think you need to have a look at Cartographer's Algorithm first. It is the Graph-based Localization with the sub-map inside. If you compare the model of both "amcl" which is occupancy grid [...]

Maybe it's possible by creating one sub map which is the occupancy values of the loaded pgm file and create sub maps on top of it? without any new trajectory nodes, what do you think? can it work?

MichaelGrupp, 2020-10-15:
There is currently no direct conversion from a map image to a Cartographer map available in this repository. And it wouldn't be trivial to do.

closed 2020-10-15 by MichaelGrupp

MigVega, 2022-11-27:
Hi all,
you can use[ this package](https://github.com/MigVega/Ogm2Pgbm/tree/main/ogm2pgbm) to do the conversion from pgm to Pose graph based map!
You don't have to install docker if you don't want. If you have already cartographer compiled you can then simply use that scripts.
Otherwise the provided instructions in the [readme](https://github.com/MigVega/Ogm2Pgbm) will allow you to compile everything in one isolated container.
Have fun, and enjoy the power of graph-based localization! :D