cartographer-project/cartographer_ros #1426: cartographer3d  gray ground and

Fanny-one, 2019-11-21:
1. Hi, @gaschler @ojura
I have try cartographer_2d and cartographer_3d  in my project. Both can produce good map, but I find the free zone or ground using cartographer_3d is gray instead white.  The first picture is from cartographer_2d and another is from cartographer_3d.
[image]
[image]
I will provide my configurefiles [here](https://github.com/Fanny-one/configurefiles).
2. Besides, when I correct IMU by [the advice](https://github.com/googlecartographer/cartographer_ros/issues/902) then set rpy 0 0 -1.63 , An error appears that odom and true direction would  have 90-degree deviation. But if I set rpy 0 0 0, the deviation appear.
[image]
Hope for you help, thans very much.

dzxbiubiubiu, 2020-05-09:
Hi:)
I have met the same question that the color of the cartographer 3D map is gray.
Do you solve it?

wjjjjyourFA, 2020-07-21:
Hi:)
I have met the same question that the color of the cartographer 3D map is gray.
when i got my lua file changed again and again, i find that white area is either the ceiling or the back of a table .
so i doubt that is there a direction of gravity or radar . but i cant test my assumption .  ..0.0..

MichaelGrupp, 2020-10-14:
You can set the transparency (alpha) of the occupancy grid in RViz settings. Then it's white.

closed 2020-10-14 by MichaelGrupp

Anthony1011, 2023-03-06:
hi everyone I meet the some issue, when I run the  cartographer 3D map, those surrounding objects will be detected as black grid but when I move the robot I can't get the withe zone (available area).
 
If I flip the imu then some grid turn white I'm still wondering why it is,
if any advice will appreciate.

[image]
[image]
[image]
.