cartographer-project/cartographer_ros #1427: Some thing wrong with my Cartographer_ros, as it could just show the fist frame and then crashed

JinTongxing, 2019-11-27:
Hi, I have just installed cartographer and cartographer ROS, as well as ceres and protoc3, but when I launch the given demo .launch file, the rviz could just show the first fame of green laser scan and the axis, then it crashed as the following infomation, could you please be kind enough to give me some suggestions? Thank you! 

`SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [16818]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 509e49ba-111a-11ea-9f62-d0abd5374e48
process[rosout-1]: started with pid [16831]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [16839]
process[cartographer_node-3]: started with pid [16849]
process[cartographer_occupancy_grid_node-4]: started with pid [16858]
process[rviz-5]: started with pid [16881]
process[playbag-6]: started with pid [16903]
`
`[ INFO] [1574861531.422735550]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1574861531.422955074]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1574861531.422984586]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1574861531.423148294]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1574861531.423185332]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1574861531.423388544]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1574861531.423413080]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1574861531.423542556]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1574861531.423565669]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1574861531.423731675]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1574861531.423758076]: I1127 21:32:11.000000 16849 configuration_file_resolver.cc:41] Found '/home/jintongxing/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1574861531.429895446]: I1127 21:32:11.000000 16849 submap_2d.cc:187] Added submap 1
[ INFO] [1574861531.429956649]: I1127 21:32:11.000000 16849 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1574861531.901067748, 1432647016.512605229]: I1127 21:32:11.000000 16849 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '635682438164904058'.
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
[cartographer_node-3] process has died [pid 16849, exit code -6, cmd /home/jintongxing/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/jintongxing/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/jintongxing/.ros/log/509e49ba-111a-11ea-9f62-d0abd5374e48/cartographer_node-3.log].
log file: /home/jintongxing/.ros/log/509e49ba-111a-11ea-9f62-d0abd5374e48/cartographer_node-3*.log
`

mrgransky, 2020-01-02:
which launch file are you trying to run?
be more specific with codes!

JinTongxing, 2020-01-02:
> which launch file are you trying to run?
> be more specific with codes!

Thanks for your reply. I just wanna run the demo as:
roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag

And in the rviz, it just looks like:
/home/jintongxing/Pictures/Screenshot from 2020-01-02 21-34-24.png

with the information in terminal:
[cartographer_node-3] process has died [pid 4434, exit code -11, cmd /home/jintongxing/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/jintongxing/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/jintongxing/.ros/log/36a36728-2d64-11ea-b468-d0abd5374e48/cartographer_node-3.log].
log file: /home/jintongxing/.ros/log/36a36728-2d64-11ea-b468-d0abd5374e48/cartographer_node-3*.log

JinTongxing, 2020-01-02:
[image]
This is the screenshoot

mrgransky, 2020-01-02:
your problem seems to start at this line:
```
terminate called after throwing an instance of 'std::out_of_range'
what(): map::at
```
maybe try to clean your `catkin_ws` once more and build it from scratch?

jxhstrive, 2020-01-07:
@JinTongxing 
Hi!
I also encountered the same problem. I tried to clean my workspace and build it from scratch. But I didn't solve it. Do you have any suggestion to solve it? Thanks!!

JinTongxing, 2020-01-21:
@jxhstrive hi, I solved it by replacing my PCL 1.9 by PCL 1.7, then it could work perfectly!

closed 2020-10-14 by MichaelGrupp