cartographer-project/cartographer_ros #1429: Source of initial guess for the pose extrapolator if no odometry and IMU data is provided?

gsh89, 2019-12-05:
Thank you for filing an issue!

It would help us tremendously if you run through the following checklist. This
ensures that we have the most information to quickly understand, analyze,
reproduce and eventually resolve your issue.

Please

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
4. remove this boilerplate text before submitting your issue.

We will likely be unable to help you without all of these points addressed.

ojura, 2019-12-06:
I believe the speed is simply estimated in `PoseExtrapolator` as the difference of the last couple of positions divided by time.

gsh89, 2019-12-06:
Thanks @ojura .  Would you also happen to know if using odometry and IMU as data sources, act only as initial guess to the scan matcher or do they affect the final pose estimate too and how can each scenario be configured ?

ojura, 2019-12-07:
The odometry and the IMU data is used to unwarp the scan (e.g. account for motion during the lidar sweep) prior to scan matching. There is nothing to configure except the use imu data/odometry settings you mentioned. Perhaps the translation and rotation weights of the scan matcher, which control how much the prior is trusted.

closed 2019-12-09 by gsh89