cartographer-project/cartographer_ros #1430: Is there an up-to-date gRPC documentation?

abylikhsanov, 2019-12-15:
Hi, 

I came across this video:
https://vimeo.com/293260413

And I decided to try this method as I am using an ICP localization technique which relies on the static objects. So after some research I found this link:
https://github.com/googlecartographer/rfcs/blob/master/text/0002-cloud-based-mapping-1.md#cartographer_grpc

And I guess it is outdated as the name of the gRPC nodes are different in the cartographer source folder. Also the official documentation includes the gRPC integration but says nothing about it. 

Is there some updated documentation about it?

shreyasgokhale, 2020-01-11:
Hi,

Have a look at https://github.com/googlecartographer/cartographer_ros/issues/1300#issuecomment-496457346 from issue #1300 . But apart from that, there is not much documentation as per my knowledge.

You can use https://github.com/shreyasgokhale/cartographer-grpc to quickly test gRPC with cartographer. 

- Shreyas

abylikhsanov, 2020-01-11:
Hi, thanks, I have already figured out. I can see that your docker compose file opens up the ports. Did you have a problem of accessing those ports of the agent is running outside the server's subnet?

shreyasgokhale, 2020-01-11:
I think you will have to tweak the IP values of files in [postinstall](https://github.com/shreyasgokhale/cartographer-grpc/tree/master/install/postinstall) folder. But as it uses docker's host networking, your agent should be able to connect to this server, (at least if it is in the same local network). If you want to connect from outside the network, you can try to use something like [this](https://github.com/jwilder/nginx-proxy).

Also, I followed [this](https://github.com/googlecartographer/cartographer_ros/issues/819#issuecomment-382003973) post to create the architecture.

Also now I look at it, I haven't updated cartographer-grpc repo in a while. I'll push my latest changes soon, with working server, agent and docs.

carola173, 2020-06-30:
@shreyasgokhale I am trying to replicate the same architecture given in the reference link. I am able to create the cartographer server node which is running properly. But when the client is create, after few seconds the client-server communication breaks with the following message. It would be great if you can help me with this error if possible.
The client side output:
[image]

The server side output:
[image]

I am running on two different Ubuntu 18.04 system (ROS Melodic) that are connected to same network in which one i am running the server and another on i am running the client. Also I am using the master repo for cartographer, cartographer_ros, asyn_grpc,grpc++(1.29) and protoc with 3 verison.

shreyasgokhale, 2020-06-30:
Hi @carola173, it is hard to debug just by looking at the error. What launch files / launch commands are you using?

carola173, 2020-07-01:
@shreyasgokhale I am using the following launch commands:
On **server side** the launch looks like : 
[image]

and configuration setup as : 
[image]

**Agent side** I am launching following stuff: 

Launch file : 
[image]

Configuration file : 
[image]

 The **cloud_slam.lua** and the **my_robot.lua** is similar to **backpack_2d.lua** file. 

I hope these are proper launch file and configuration. Thanks a lot for helping.

MichaelGrupp, 2020-10-14:
My older answere here is still mostly valid: https://github.com/cartographer-project/cartographer_ros/issues/1300#issuecomment-496457346

The project is being maintained again, but I can't tell you if, when or how an extended public documentation of the cloud features will happen.

closed 2020-10-14 by MichaelGrupp