cartographer-project/cartographer_ros #1431: Correct tf-order for odom frame when use_odometry=true

TillmanOttBA, 2019-12-17:
Hi all,

I am currently trying to improve the accuracy of the cartographer by using odometry. However, the first result was not very satisfying as the cartographer automatically places the origin of the odom frame directly into the origin of the base_link frame. My first attempt was to create the odom-frame in the robot model and link it to the base_link with the correct frame offset:
[image]
[image]
```
<robot name="broetje_Prototype_AGV">

  <material name="gray">
    <color rgba="0.7 0.7 0.7 0.5" />
  </material>
  <material name="red">
    <color rgba="1.0 0.0 0.0 1.0" />
  </material>
  <material name="yellow">
    <color rgba="1.0 0.73 0.0 0.5" />
  </material>

  <!--LINK: Map-->
  <!--link name="map">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
    </visual>
  </link-->
  <!--JOINT: Odometry-->
  <!--joint name="map_to_odom" type="fixed">
    <parent link="map" />
    <child link="odom" />
    <origin rpy="0 0 0" xyz="0.5 -0.3 0"/>
  </joint-->

  <!--LINK: Odometry-->
  <!--link name="odom">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
    </visual>
  </link-->
  <!--JOINT: Base-->
  <!--joint name="odom_to_base" type="fixed">
    <parent link="odom" />
    <child link="base_link" />
    <origin rpy="0 0 0" xyz="-0.5 0.3 0"/>
  </joint-->

  <!--LINK: Base-->
  <link name="base_link">
    <visual>
      <geometry>
        <mesh filename="package://cartographer_ros/meshes/Cobot2019_origin.dae"/>
        <!--box size="1 0.6 0.45"/-->
      </geometry>
      <origin rpy="0 0 0" xyz="0.5 -0.3 0"/>
      <material name="red"/>
    </visual>
  </link>
  <!--JOINT: Safety Scanner 1-->
  <joint name="base_to_scan_1" type="fixed">
    <parent link="base_link" />
    <child link="scan_1" />
    <origin rpy="3.1416 0 2.3561925" xyz="0.055 -0.055 0.18"/>
  </joint>
  <!--JOINT: Safety Scanner 2-->
  <joint name="base_to_scan_2" type="fixed">
    <parent link="base_link" />
    <child link="scan_2" />
    <origin rpy="3.1416 0 -0.7853975" xyz="0.956 -0.556 0.18"/>
  </joint>
  <!--JOINT: Odometry-->
  <joint name="base_to_odom" type="fixed">
    <parent link="base_link" />
    <child link="odom" />
    <origin rpy="0 0 0" xyz="0.5 -0.3 0"/>
  </joint>

  <!--LINK: Safety Scanner 1-->
  <link name="scan_1">
    <visual>
      <geometry>
        <mesh filename="package://cartographer_ros/meshes/sick_scanner_origin.dae"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <material name="yellow"/>
    </visual>
  </link>

  <!--LINK: Safety Scanner 2-->
  <link name="scan_2">
    <visual>
      <geometry>  
        <mesh filename="package://cartographer_ros/meshes/sick_scanner_origin.dae"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <material name="yellow"/>
    </visual>
  </link>

    <!--LINK: Odometry-->
  <link name="odom">
    <visual>
      <origin rpy="0 0 0" xyz="0.5 -0.3 0"/>
    </visual>
  </link>

</robot>
```
If I proceed as displayed above, base_link or odom starts jumping around while rotating the AGV. The jumping is not due to the odometry itself. It works with low latency and is quite accurate. Outside the cartographer there are no problems. Also it is sometimes indicated that odom causes a closed-loop, because it seems to be parent of base_link as well. This might be the course of the problem.

I tried to move the odom-frame between the map and the base_link as the cartographer does when the publish_odometrie=true option is selected. This of course leads to the odom-frame not moving anymore, because then it has a fixed connection to the map-frame.

At this point, I don't really know how to proceed. Maybe one of you can tell me where the odom frame has to be linked. If there is any information missing, please let me know.

Thanks a lot 
Tillman

(I also asked this question [on Ros-Answers](https://answers.ros.org/question/340041/correct-tf-order-for-odom-frame-in-cartographer/), but I think I could be more successful with this question here)

abylikhsanov, 2019-12-22:
Your TF tree is incorrect. You do not need to create the Odom frame in the URDF. Cartographer can provide that TF using the config file

MichaelGrupp, 2020-10-14:
The order should be map -> odom -> base_link. map -> odom is provided by Cartographer. odom -> base_link can be provided either by an external odometry or by Cartographer itself (see the documentation, `provide_odom_frame` parameter).

map, odom and base_link line up at the very beginning. This is expected behavior.

closed 2020-10-14 by MichaelGrupp

kmlingaudhaya, 2024-01-16:
> The order should be map -> odom -> base_link. map -> odom is provided by Cartographer. odom -> base_link can be provided either by an external odometry or by Cartographer itself (see the [...]

I have a question. 
If odom to base link transform can be calculated using imu and wheel encoders. then by saying map->odom, what are we implying?
1) Are we saying map frame is fixed? If so then odom is fixed or not?
2) Odom to base link can be calculated through wheel encoders and imu as mentioned earlier. Then how the transform from map to odom is computed, because odom frame is not moving anywhere, while we have base_link moving with respect to odom.

how this architecture helps us in rectifying the drift in odom to base_link too?
I hope my question and doubt is clear.