cartographer-project/cartographer_ros #1435: increasing resolution for mapping

altineller, 2020-01-08:
Hello,

What can I do to increase the resolution of the cartographer mapping? In gmapping or karto-slam there is an option to set the resolution, but for cartographer I could not find. 

Which set of parameters are responsible for resolution in cartographer?

Best regards,
C.

Qiuxiaochang, 2020-01-09:
3D SLAM, you can find it in cartographer/configuration_files/trajectory_builder_3d.lua. 2D SLAM is similar.

nerellasureshkumar7, 2020-05-24:
Hi @altineller , we can change the resolution of the cartographer map by passing the resolution argument to the _cartographer_occupancy_grid_node_. By default it is 0.05.

`<node name="cartographer_occupancy_grid_node" pkg="cartographer_ros" type="cartographer_occupancy_grid_node" args="-resolution 0.05" />`

MichaelGrupp, 2020-10-14:
@Qiuxiaochang is right. You need to change the parameter in your Lua config. The occupancy grid node mentioned by @nerellasureshkumar7 is a separate process that has no effect on the internal resolution of SLAM.

closed 2020-10-14 by MichaelGrupp

marcoluis97, 2023-01-26:
Hi @MichaelGrupp @Qiuxiaochang! After setting a resolution of 0.005, which is an order of magnitude higher of the default resolution, the time needed in order to build an optimize the map seems absurdly long. Am I limited by the processing capacity of my cpu, or are there other parameters I must adjust in order to achieve a desired higher map resolution?

MichaelGrupp, 2023-01-26:
If you're using 2D SLAM, 0.05 to 0.005 means 100 times more data in the grid. I would say that it's expected that this has an impact on performance. Probably a dense grid-based mapping approach like cartographer is doing in 2D is not the ideal choice for such settings.

marcoluis97, 2023-01-26:
Makes sense, thank you @MichaelGrupp! I've come to realize that for a more realistic value of resolution, I should aim for values between 0.05 and 0.01. I would like to ask you another question if possible. Besides wanting to up the resolution of my occupancy grid, I am also looking to achieve a better representation than what I've got so far. The full issue is described in #1763 but basically after setting a resolution of 0.01 the map result is not yet globally consistent. There are still some artifacts and double walls appearing. If you could give me any idea of what I could do to find a solution or point me in direction to get answers I would be very thankful!