cartographer-project/cartographer_ros #1437: Question: Convert LatLng to ECEF, then to odom.

hngu97, 2020-01-16:
Hi, I have used robot_localization and move_base in ROS before, in which I used TF from robot_localization (utm->map->odom->base_link) to transform the goal points from UTM to odom frame and published these goals for the robot to follow. Is this possible to do using TF from Cartographer? I know that Cartographer using ECEF but I am not really sure. I believe there is some helper functions to help me convert LatLng to ECEF, but how can I convert ECEF points to odom given that cartographer only publish map->odom->base_link? Thank you.

soorajanilkumar, 2020-01-31:
Hi,
I was also looking at the same thing, just found one that might work. 
Maybe this will library will help you convert LatLng to local cartesian coordinates. 
[GeographicLib::LocalCartesian Class Reference](https://geographiclib.sourceforge.io/html/classGeographicLib_1_1LocalCartesian.html#details)

There are other functions in the library that might be useful too.

MichaelGrupp, 2020-10-14:
Not sure if this here can also help you: https://github.com/cartographer-project/cartographer_ros/blob/8bbec6bff77a7e63669db34b9023ad1811e741fd/cartographer_ros/cartographer_ros/msg_conversion.h#L77

Closing as answered.

closed 2020-10-14 by MichaelGrupp