cartographer-project/cartographer_ros #1443: Obtain occupancy grid map from localization mode trajectory submap

soorajanilkumar, 2020-02-05:
Hi cartographer community,

Is there a way to convert the submaps associated with the trajectory 1 (or current traj) in pure localization mode to occupancy grid map and publish it? 
I would like to obtain the constantly updated and trimmed submap when running the localization mode and require it in the form of occupancy grid map.

Any help or suggestion would be very helpful for me. 

Thank you all in advance
Sooraj

MichaelGrupp, 2020-10-14:
You can run the occupancy grid node and pass `--include_frozen_submaps=false` as an argument. Then the static frozen map won't be included and you should only see your active trajectory.
https://github.com/cartographer-project/cartographer_ros/blob/8064b8f9b71fde9e375d92321e5e307a96b45f07/cartographer_ros/cartographer_ros/occupancy_grid_node_main.cc#L43

closed 2020-10-14 by MichaelGrupp

soorajanilkumar, 2020-10-16:
T

> You can run the occupancy grid node and pass `--include_frozen_submaps=false` as an argument. Then the static frozen map won't be included and you should only see your active trajectory. [...]

Thanks a lot. Didn't know the option existed! 
It does exactly what I want.