cartographer-project/cartographer_ros #1444: Wired tf jump behavior

pl95, 2020-02-05:
Hi, 
I have a wired tf jump behavior when I launch the mapping file. The tf of the upper body of the robot jump around when the base of the robot walking normally as the following gif shows.
Everything else work normally. Is there any way I can fix this?

[image]

1. [rosbag_validate_result](https://gist.github.com/pl95/ce5153e74fd3b8bb8e0eb2d7f249e303):  
2. [ Launch, lua configuration and URDF
](https://github.com/pl95/cartographer_issue)
3. My bag file:  [joytf.bag](https://drive.google.com/open?id=1ldsdDftrWgNhfUPbmycvMET-pMLkxiM-)
The data is record from real robot. 

Thank you so much!!

qvntm, 2020-02-12:
You commented out robot_state_publisher? That publishes tf between robot links so probably that's the issue. And since its tf issue, its better to provide tf info as well. You can get the pdf by `rosrun tf view_frames` while running the nodes.

mocibb, 2020-02-27:
There is not problem in the bag provided. I can run this bag with very good result.
[image]

closed 2020-10-14 by MichaelGrupp

unirlm, 2022-12-20:
how did you solve this issue?