cartographer-project/cartographer_ros #1446: Is it possible to switch between slam and self-localization mode at runtime

maxbader, 2020-02-24:
I have recorded a pbstream and I can use it to localize myself (2D).
I would like to change the parameters
```
TRAJECTORY_BUILDER.pure_localization = true
POSE_GRAPH.optimize_every_n_nodes = 20
```
at runtime to switch back to SLAM mode. 
Is this possible using ROS?

artemiialessandrini, 2020-04-24:
If I'm not mistaken, in ROS it's usually done using [dynamic_reconfigure](wiki.ros.org/dynamic_reconfigure). Not sure how to link it with cartographer_ros though.

shaoxiang, 2020-06-15:
@maxbader Hi, I have the same need. Have you solved it? 
I'm trying to see if I can change the Lua configuration dynamically.

MichaelGrupp, 2020-09-01:
You can use the /start_trajectory service to start a trajectory with a different configuration. Note that this only changes the trajectory builder options, map builder / pose graph options can't be changed during runtime. Enabling / disabling trimming works though.

closed 2020-10-14 by MichaelGrupp