cartographer-project/cartographer_ros #1447: change /odom topic

alexsimon1, 2020-02-24:
hello,

is it possible to change the expected /odom topic when i use odometry?

Thank You

nerellasureshkumar7, 2020-03-20:
hi,
yes it is possible to change the odometry topic using **remap ** tag in launch file.

`     <node name="cartographer_node" pkg="cartographer_ros" type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename robot.lua">
           <remap from="odom" to="wheel_odom" />
  </node>
`

nyangshawbin, 2020-05-23:
Hi @ojura , 
I am similarly looking at changing the `/odom` topic expected by Cartographer. 
I am running Cartographer offline on a rosbag recorded from Gazebo (turtlebot3). However, prior to running Cartographer, I made amendments to the original `/odom` topic to include noise and have this noisy odom data published to a new topic `/odom_noise`, all of which were recorded in a rosbag. 

As such, by specifying `use_odometry = true` in Cartographer's lua config file, I want Cartographer to take this in noisy `/odom_noise` topic as odometry input. 

This is a snippet of my lua config file:
`map_frame = "map",`
`tracking_frame = "base_footprint", --use base_footprint if working with turtlebot in gazebo`
`published_frame = "odom",`  
`odom_frame = "odom",`
`provide_odom_frame = false, `
`use_odometry = true,`

Prior to remapping `/odom` into `/odom_noise` as recommended by @nerellasureshkumar7, my tf_tree looks like this, with cartpgrapher still being able to run and mapping to be done in rviz:
[image]

However, after remapping the odom topics, i encountered the error repeatedly:
`[ WARN] [1590222644.262712405, 325.915000000]: W0523 16:30:44.000000 24677 tf_bridge.cc:52] Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_co`

[image]

I would really appreciate it someone can give me some advice on this. Thanks in advance!

nerellasureshkumar7, 2020-05-23:
Hi @nyangshawbin 
**remapping** the topic name will work.If you could share the launch file and lua file it would be helpful. I hope you have remapped the topic name perfectly.

nyangshawbin, 2020-05-23:
Hi,
Following are the files:
[backpack_3d.lua.pdf](https://github.com/cartographer-project/cartographer_ros/files/4671578/backpack_3d.lua.pdf)
[backpack_3d.launch.pdf](https://github.com/cartographer-project/cartographer_ros/files/4671579/backpack_3d.launch.pdf)

I made some changes to the way i published the odom noise and this is my new tf tree:

[image]
Topics in my rosbag:

[image]

Thank you!

nerellasureshkumar7, 2020-05-23:
Everything looks fine to me, Please send the rosbag file. I will check once.

nyangshawbin, 2020-05-23:
Heres to the [rosbag](https://drive.google.com/file/d/1l3hbknQoo_07scx2QtB1k5T39jkDjg4L/view?usp=sharing) stored on gdrive temporarily. Thank you so much. I will let you know should i manage to get it running!

nerellasureshkumar7, 2020-05-23:
Hi @nyangshawbin 
I have checked bag file it seems there is an issue with odom_noise. 
In  _odom_noise_ there is no **child_frame_id**. 
[image]
Add the **child_frame_id** as **base_footprint** it will work.

nyangshawbin, 2020-05-23:
Hi @nerellasureshkumar7, it works! Overlook the need to map over the child_id when publishing to a new topic. Huge thank you for your guidance, really appreciate it! Cheers.

bochen87, 2020-06-17:
Closing since resolved

closed 2020-06-17 by bochen87