cartographer-project/cartographer_ros #1452: Error when building cartographer rviz module in final command in install guide

jacksonjnh34, 2020-03-08:
When attempting to build cartographer_ros I get a decent way through the entire process, but when making cartographer_rviz, I get a plethora of PCL errors.

I don't know if this is because I have PCL 1.10 installed (which would be surprising) but here is the start of the error:

`[3/4] Building CXX object CMakeFiles/cartographer_rviz.dir/cartographer_rviz/drawable_submap.cc.o
FAILED: CMakeFiles/cartographer_rviz.dir/cartographer_rviz/drawable_submap.cc.o 
/usr/bin/c++  -DGFLAGS_IS_A_DLL=0 -DQT_NO_KEYWORDS -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"cartographer_rviz\" -Dcartographer_rviz_EXPORTS -Icartographer_rviz_autogen/include -I/usr/include/aarch64-linux-gnu/qt5 -I/usr/include/aarch64-linux-gnu/qt5/QtWidgets -I/usr/include/aarch64-linux-gnu/qt5/QtGui -I/usr/include/aarch64-linux-gnu/qt5/QtCore -I/usr/lib/aarch64-linux-gnu/qt5/mkspecs/linux-g++ -I. -I/home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz -isystem /usr/include/eigen3 -isystem /home/kart/catkin_ws/install_isolated/include -isystem /opt/ros/melodic/include -isystem /opt/ros/melodic/share/orocos_kdl/cmake/../../../include -isystem /opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/local/include/pcl-1.10 -isystem /usr/include/pcl-1.8 -isystem /usr/include/ni -isystem /usr/include/openni2 -isystem /usr/include/vtk-6.3 -isystem /usr/include/freetype2 -isystem /usr/lib/aarch64-linux-gnu/openmpi/include/openmpi -isystem /usr/lib/aarch64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent -isystem /usr/lib/aarch64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include -isystem /usr/lib/aarch64-linux-gnu/openmpi/include -isystem /usr/include/python2.7 -isystem /usr/include/aarch64-linux-gnu -isystem /usr/include/hdf5/openmpi -isystem /usr/include/libxml2 -isystem /usr/include/jsoncpp -isystem /usr/include/tcl -isystem /usr/include/OGRE/Overlay -isystem /usr/include/OGRE -isystem /usr/include/lua5.2 -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/aarch64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/libpng16 -isystem /usr/local/include -O3 -DNDEBUG -fPIC    -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MD -MT CMakeFiles/cartographer_rviz.dir/cartographer_rviz/drawable_submap.cc.o -MF CMakeFiles/cartographer_rviz.dir/cartographer_rviz/drawable_submap.cc.o.d -o CMakeFiles/cartographer_rviz.dir/cartographer_rviz/drawable_submap.cc.o -c /home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.cc
In file included from /usr/local/include/pcl-1.10/pcl/pcl_macros.h:77:0,
                 from /usr/local/include/pcl-1.10/pcl/PCLHeader.h:10,
                 from /usr/local/include/pcl-1.10/pcl/point_cloud.h:47,
                 from /home/kart/catkin_ws/install_isolated/include/cartographer_ros/msg_conversion.h:31,
                 from /home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.cc:28:
/usr/local/include/pcl-1.10/pcl/pcl_config.h:7:4: error: #error PCL requires C++14 or above
   #error PCL requires C++14 or above
    ^~~~~
In file included from /usr/local/include/pcl-1.10/pcl/console/print.h:44:0,
                 from /usr/local/include/pcl-1.10/pcl/conversions.h:53,
                 from /opt/ros/melodic/include/pcl_conversions/pcl_conversions.h:44,
                 from /home/kart/catkin_ws/install_isolated/include/cartographer_ros/msg_conversion.h:33,
                 from /home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.cc:28:
/usr/local/include/pcl-1.10/pcl/pcl_config.h:7:4: error: #error PCL requires C++14 or above
   #error PCL requires C++14 or above`

There are literally thousands and thousands of these PCL error lines, so I won't be pasting them all. The final error message is as follows:

`In file included from /usr/include/c++/7/bits/stl_algobase.h:71:0,
                 from /usr/include/c++/7/bits/char_traits.h:39,
                 from /usr/include/c++/7/string:40,
                 from /usr/include/c++/7/stdexcept:39,
                 from /usr/include/c++/7/array:39,
                 from /usr/include/c++/7/tuple:39,
                 from /usr/include/c++/7/mutex:38,
                 from /usr/include/c++/7/future:38,
                 from /home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.h:20,
                 from /home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.cc:17:
/usr/include/c++/7/bits/predefined_ops.h: In instantiation of ‘bool __gnu_cxx::__ops::_Iter_pred<_Predicate>::operator()(_Iterator) [with _Iterator = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>]’:
/usr/include/c++/7/bits/stl_algo.h:120:14:   required from ‘_RandomAccessIterator std::__find_if(_RandomAccessIterator, _RandomAccessIterator, _Predicate, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)> >]’
/usr/include/c++/7/bits/stl_algo.h:161:23:   required from ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)> >]’
/usr/include/c++/7/bits/stl_algo.h:3932:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<const pcl::PCLPointField*, std::vector<pcl::PCLPointField> >; _Predicate = pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>]’
/usr/local/include/pcl-1.10/pcl/common/io.h:65:77:   required from here
/usr/include/c++/7/bits/predefined_ops.h:283:11: error: no match for call to ‘(pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>) (const pcl::PCLPointField&)’
  { return bool(_M_pred(*__it)); }
           ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/pcl-1.10/pcl/io/file_io.h:41:0,
                 from /usr/local/include/pcl-1.10/pcl/io/pcd_io.h:44,
                 from /opt/ros/melodic/include/pcl_conversions/pcl_conversions.h:70,
                 from /home/kart/catkin_ws/install_isolated/include/cartographer_ros/msg_conversion.h:33,
                 from /home/kart/catkin_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.cc:28:
/usr/local/include/pcl-1.10/pcl/common/io.h:65:39: note: candidate: pcl::getFieldIndex(const pcl::PCLPointCloud2&, const string&)::<lambda(int)>
         [&field_name](const auto field) { return field.name == field_name; });
                                       ^
/usr/local/include/pcl-1.10/pcl/common/io.h:65:39: note:   no known conversion for argument 1 from ‘const pcl::PCLPointField’ to ‘int’
ninja: build stopped: subcommand failed.`

I would note that I am not actually running the cartographer full catkin_make command for this, rather the command generated by the install script that tells me how to recreate the error. Any help would be greatly appreciated!

HTLife, 2020-06-26:
Same issue here.   @jacksonjnh34 any solution after then?

MichaelGrupp, 2020-10-14:
It requires C++14, as written in the error log. We fixed the CMake code of Cartographer recently so this shouldn't happen anymore.

closed 2020-10-14 by MichaelGrupp