cartographer-project/cartographer_ros #1454: Can't be loop,please help

edwardlilqqs, 2020-03-12:
Thank you for filing an issue!

It would help us tremendously if you run through the following checklist. This
ensures that we have the most information to quickly understand, analyze,
reproduce and eventually resolve your issue.

Please
[image]

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
4. remove this boilerplate text before submitting your issue.

We will likely be unable to help you without all of these points addressed.

edwardlilqqs, 2020-03-12:
how to tune parameter

artemiialessandrini, 2020-04-15:
Hi, I'd take a look [here](https://github.com/cartographer-project/cartographer_ros/commit/698fc1043fb6c59d84a70cad1fa79b99a385dfee) first, especially about ```ceres_scan_matcher```

MichaelGrupp, 2020-10-14:
Link to the latest version of the tuning guide: https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html