cartographer-project/cartographer_ros #1456: Can some One provide a pipline to run the cartographer using Odom ? (Thanks, almost providing the rosbag)

qpc001, 2020-03-14:
Can some One provide a pipline to run the cartographer using Odom ? (Thanks, almost providing the rosbag)

numb0824, 2020-08-04:
I have a problem with running cartographer using odom and laser,could help me that how should i write the lua?

3473f, 2020-09-13:
@numb0824  That's the .lua file we are using for our robot with a SICK S300 laser scanner and odometry
`include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = false
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 15
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 10

POSE_GRAPH.optimization_problem.local_slam_pose_translation_weight = 10
POSE_GRAPH.optimization_problem.local_slam_pose_rotation_weight = 10
POSE_GRAPH.optimization_problem.odometry_translation_weight = 5
POSE_GRAPH.optimization_problem.odometry_rotation_weight = 0.01

return options`

You are going to have to tune the parameters to suit how much you trust the data sources.

closed 2020-10-14 by MichaelGrupp