cartographer-project/cartographer_ros #1459: rviz Initiating shutdown! during cartographer_ros

SeungHunJeon, 2020-03-25:
I used cartographer with IMU, Lidar sensors It goes well. But after running 10minutes, these codes occurs and rviz died.

How can I deal with this problem?

[ INFO] [1585114410.974068648]: I0325 14:33:30.000000 19238 submap_3d.cc:321] Added submap 52
[ INFO] [1585114412.701197439]: I0325 14:33:32.000000 19238 collated_trajectory_builder.cc:72] imu rate: 50.00 Hz 2.00e-02 s +/- 2.36e-03 s (pulsed at 100.31% real time)
[ INFO] [1585114412.701239193]: I0325 14:33:32.000000 19238 collated_trajectory_builder.cc:72] points2 rate: 20.01 Hz 5.00e-02 s +/- 1.74e-03 s (pulsed at 100.64% real time)
[ INFO] [1585114413.540777029]: I0325 14:33:33.000000 19238 motion_filter.cc:42] Motion filter reduced the number of nodes to 56.4828%.
[ INFO] [1585114440.092952119]: I0325 14:34:00.000000 19238 collated_trajectory_builder.cc:72] imu rate: 46.56 Hz 2.15e-02 s +/- -nan s (pulsed at 0.13% real time)
[ INFO] [1585114440.093069190]: I0325 14:34:00.000000 19238 collated_trajectory_builder.cc:72] points2 rate: 20.00 Hz 5.00e-02 s +/- 1.70e-03 s (pulsed at 100.13% real time)
================================================================================REQUIRED process [rviz-4] has died!
process has died [pid 19240, exit code -9, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/jsh/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/demo_3d.rviz __name:=rviz __log:=/home/jsh/.ros/log/85ef58b4-6d97-11ea-9bdf-704d7b8a20bd/rviz-4.log].
log file: /home/jsh/.ros/log/85ef58b4-6d97-11ea-9bdf-704d7b8a20bd/rviz-4*.log
Initiating shutdown!
================================================================================
[ INFO] [1585114440.498796034]: I0325 14:34:00.000000 19238 submap_3d.cc:321] Added submap 53
[playbag-5] killing on exit
[rviz-4] killing on exit
[cartographer_occupancy_grid_node-3] killing on exit
[cartographer_node-2] killing on exit
[robot_state_publisher-1] killing on exit
Optimizing: 100.0%...[cartographer_node-2] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete

MichaelGrupp, 2020-10-14:
This looks like a problem of RViz, not Cartographer. You might need to debug what happens exactly.

closed 2020-10-14 by MichaelGrupp

Ali-Darzi, 2022-06-14:
I ran the cartographer_ros in docker image and got the same issue. The problem is fixed doing two steps:
1- Installing the latest version of graphic card driver
2- Mounting /tmp/.X11-unix to the docker image
docker run -it --net=host \
    --env="DISPLAY" \
    --env="QT_X11_NO_MITSHM=1" \
    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    --volume="/home/Downloads:/home/Downloads"\
    cartographer_ros