cartographer-project/cartographer_ros #1460: How to georeference Cartographer output

bj-neilson, 2020-03-30:
Hi,

Is it possible to add geo location data such as projection and datum to .ply pointcloud generated by Cartographer? 

Or should I sort of work backwards by adding Landmarks to Cartographer to use as Ground Control Points?

Any other suggestions please?

Ta,
Ben

Soini3913, 2020-04-30:
Hi Ben,

as it seems google isn`t supporting it anymore. Here is my way of "georefrencing" the map. if you increase the weight of the fixed frame pose, you can achieve a georeferenced map (the starting point is printed on the console and form there everything is metric system). 
But the problem there is that the pointcloud matching does not have a high influence...maybe we can work out something together.

Best,
Sven

MichaelGrupp, 2020-10-21:
This PLY annotation is not supported, and it also sounds like a special use case that might be more suitable to be addressed in a fork. We do support GPS input though, but this is probably not what you mean.

closed 2020-10-21 by MichaelGrupp