cartographer-project/cartographer_ros #1465: Error when using /start_trajectory service

wuchenxi1996, 2020-04-11:
I used pure localiztion mode of cartographer and I finished the last trajectory 1 with rosservice call /finish_trajectory 1. Then I wanted to start  a new trajectory by using /start_trajectory service with arguments, but the feedback I got showed: 
ERROR: transport error completing service call: unable to receive data from sender. 
    Does anyone encountes the same problem and know how to fix it?

MichaelGrupp, 2020-09-01:
Looks like either a ROS communication problem or a crash of Cartographer, can't really tell. I don't encounter this problem with a recent version of Cartographer and we are heavily using these services.

closed 2020-10-21 by MichaelGrupp

skywno, 2022-04-22:
I actually encountered the same problem.  It only happens when I call /start_trajectory service.
[image]