cartographer-project/cartographer_ros #1467: To few poses

alexsimon1, 2020-04-16:
Hello,

i have much fewer Poses (picture1) with cartographer  as with hector slam (picture2), even if the motion filter has the same parameter in both slams.

[image]

[image]

Is there a workaround for this Situation?

Thanks.

MichaelGrupp, 2020-09-01:
Please take a look at: https://github.com/cartographer-project/cartographer_ros/issues/1374

closed 2020-10-14 by MichaelGrupp

gaussiangit, 2021-04-15:
@MichaelGrupp I have another camera along with lidar. Number of camera images and poses/node_index should match in my case! How can I achieve that ?