cartographer-project/cartographer_ros #1468: A request on how to use QR-code with cartographer

JiyongOh, 2020-04-20:
Hi, everyone.

I want to use cartographer for SLAM and localization under the environment that QR-codes are attached on floor in a form of grid.

I’m supposed to use the package of 'visp_auto_tracker' to detect the QR-codes and Turtlebot3 as a mobile platform which is equipped with an additional camera sensor for QR-code recognition.

I found some issues on the use of landmarks such as #909 and #1067.

Nonetheless, I can not utilize the carographer_ros package with QR-codes attached on floor.
According to the cartographer manual, it is necessary to set ‘use_landmarks = true’ in a .lua configuration file to publishing another topic related to a landmark.

However, I wonder how to define the message corresponding to the QR-code detected by visp_auto_tracker and how to assign the (global or local) coordinates to a detected QR-code.

Could you give me a guide in detail?

Please help me.

ojura, 2020-04-26:
Hi @JiyongOh, I can give you some short hints: the Landmark functionality in Cartographer optimises the trajectory according to observations of landmarks from the vehicle. So when you feed Cartographer with `LandmarkObservations`, you need to include the time of the observation (because this determines the part of the trajectory where the observation was made), and the relative pose of the landmark i.e. QR code as observed from the vehicle frame. If you only have the relative position of the landmark, you can ignore the landmark orientation by setting the landmark rotation weight to zero.

MichaelGrupp, 2020-09-01:
There is a testing script that publishes fake landmarks that are compatible with Cartographer. You can have a look at how it publishes data and use it to check if Cartographer is configured correctly to accept these landmarks.

https://github.com/cartographer-project/cartographer_ros/blob/master/cartographer_ros/scripts/dev/publish_fake_random_landmarks.py

closed 2020-10-14 by MichaelGrupp