cartographer-project/cartographer_ros #1469: Ubuntu 19.10 building

tomlogan501, 2020-04-23:
Hi,
I try to compile from source the package on my computer Ubuntu 19.10
I tried before the docker xenial without a problem.

I got an error I think is based on my gcc version 9.
```
gcc --version
gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
```
The error
```
Scanning dependencies of target cartographer.mapping.3d.hybrid_grid_test
[ 94%] Building CXX object CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o
In file included from /usr/include/c++/9/map:61,
                 from /home/tsimon/carto_ws/devel_isolated/ceres-solver/include/ceres/ordered_groups.h:34,
                 from /home/tsimon/carto_ws/devel_isolated/ceres-solver/include/ceres/ceres.h:52,
                 from /home/tsimon/carto_ws/src/cartographer/cartographer/common/math.h:25,
                 from /home/tsimon/carto_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid.h:28,
                 from /home/tsimon/carto_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:17:
/usr/include/c++/9/bits/stl_map.h: In instantiation of ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’:
/home/tsimon/carto_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9/bits/stl_map.h:273:4: error: no matching function for call to ‘std::_Rb_tree<Eigen::Matrix<int, 3, 1>, std::pair<const Eigen::Matrix<int, 3, 1>, float>, std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >, cartographer::mapping::{anonymous}::EigenComparator, std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> > >::_M_insert_range_unique(cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator&, cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator&)’
  273 |  { _M_t._M_insert_range_unique(__first, __last); }
```
What version should I install as alternatives to achieve the compilation ?

Thanks

bochen87, 2020-06-17:
Hi, I don’t think there is a ROS Version supporting Ubuntu 19. You should try to use with Ubuntu 18 or 20.

closed 2020-06-17 by bochen87

tomlogan501, 2020-06-18:
I will switch to ubuntu 20, but that's not a correct answer, I commonly use melodic on ubuntu 19 without problem, from source.
I will try with previous versions of gcc such as gcc-8