cartographer-project/cartographer_ros #1473: Reduce inter constraint (loop closure) search distance?

soorajanilkumar, 2020-05-13:
Hi,

I'm having trouble localizing in the map between very two similar corridors. The global SLAM detects and builds a faulty inter constraint with the opposite corridor. 
While mapping I was able to increase the min score for the constraint builder. But it is not viable for localization mode.
Is there a way to limit loop closure search distance, so that global or inter constraints are only searched for nodes within a threshold distance of the current pose?

Thanks in advance.

MichaelGrupp, 2020-09-01:
Did you have a look at the available Lua options? E.g.:
```
POSE_GRAPH.constraint_builder.max_constraint_distance
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.angular_search_window
```

closed 2020-10-14 by MichaelGrupp

Astrophilil, 2022-08-24:
Hello, I also have similar problems. Can you solve them? I tried to change the parameters you mentioned. After several tests, I found that it was not very effective for pure positioning correction

soorajanilkumar, 2022-09-14:
@Astrophilil 
There are a couple of things I tried.

First is to disable global sampling to only check for constraints near by. Also increased global-search-after-seconds to very high value
```
global_sampling_ratio = 0
global_constraint_search_after_n_seconds = 1e9
```
After disabling global search, you can try to play with the parameters as mentioned by Michael. Reducing it to not search beyond a limit.
```
POSE_GRAPH.constraint_builder.max_constraint_distance
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.angular_search_window
```

If even after doing the changes, u see that there are false matches, you can try to reduce (or restrict more) the huberscale until you weed out all incorrect constraints represented by blue rviz markers (i think interconstraints residues).
`POSE_GRAPH.optimization_problem.huber_scale = 10`

Maybe these steps would help u out.