cartographer-project/cartographer_ros #1476: In-place rotation localization continious jumps, that leads to navigation issues

artemiialessandrini, 2020-05-23:
I'm trying to use the cartographer's pure_localization for navigation, but it seems like it has big issues with localizing while doing in-place rotation or any others (find gif below):
[image]

I'm attaching my pure_localization.lua file

```
include "iron_car.lua"

TRAJECTORY_BUILDER.pure_localization_trimmer = {
  max_submaps_to_keep = 3,
  }

POSE_GRAPH.optimize_every_n_nodes = 1
POSE_GRAPH.constraint_builder.sampling_ratio = 0.01
POSE_GRAPH.global_sampling_ratio = 0.005
MAP_BUILDER.num_background_threads = 16

POSE_GRAPH.global_constraint_search_after_n_seconds = 8.0
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 16
POSE_GRAPH.constraint_builder.min_score = 0.45

POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window = 1.75
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.angular_search_window = math.rad(30.0)

return options
```
Also [here](https://gist.github.com/artemiialessandrini/b975f69b09541989ec136efdd22805ca) you can find a regular .lua file.

I'm using imu and odometry, lidar is Hokuyo (40hz).

It seems like I already achieved Low latency mode for localization purposes, 
so I suspect there's something noisy with sensors, but have no idea how to debug it.

Any help or ideas would be appreciated, thanks!

jonbinney, 2020-07-14:
It's also possible that rviz is exaggerating the problem. Are you synchronizing the rviz graphics to the laser data? The option will appear if you check "Experimental" at the bottom of the rviz window:
[image]

artemiialessandrini, 2020-08-11:
@jonbinney sorry for the late response and thank you for answering!

After some research and tuning, it seems like the problem was in my rotation odometry precision (see gif below)  

[image]
 

While set `use_odometry = false`, jumps disappeared.

So based on [odometry-in-global-optimization tuning](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#odometry-in-global-optimization) and [#770 ](https://github.com/cartographer-project/cartographer_ros/issues/770), solution is to descrease rotation weight for odometry in the optimization:
```
POSE_GRAPH.optimization_problem.odometry_translation_weight = 1e5 -- 1e5
POSE_GRAPH.optimization_problem.odometry_rotation_weight = 1e4 -- <--solves odom rotation error  -- 1e5
```
Hope it will be useful for others!

closed 2020-10-26 by MichaelGrupp